open62541/open62541 #5065: Memory leak UA_ClientConfig_setDefault on windows with OpenSSL

gha88, 2022-04-15:
## Description
Using a testing environment on windows 10 with open62451 version 1.2 compiled with openssl, I come up sometimes with some memory leaks.

During my investigation the problem seems to be related on the function UA_SecurityPolicy_None called by UA_ClientConfig_setDefault.

In particular in the callstack the allocated memory should be at UA_OpenSSL_LoadPemCertificate:
`bio = BIO_new_mem_buf((const void *) certificate->data, (int) certificate->length);`

I don't understand why in UA_SecurityPolicy_None, even if I won't use any encryption the software  call UA_OpenSSL_LoadLocalCertificate(&localCertificate, &policy->localCertificate);

In this case &policy->localCertificate is null.

## Background Information / Reproduction Steps
Calling UA_ClientConfig_setDefault(cc) and later UA_SecurityPolicy_None used this part of software:

`
#ifdef UA_ENABLE_ENCRYPTION_MBEDTLS
    UA_mbedTLS_LoadLocalCertificate(&localCertificate, &policy->localCertificate);
#elif defined(UA_ENABLE_ENCRYPTION_OPENSSL)
    UA_OpenSSL_LoadLocalCertificate(&localCertificate, &policy->localCertificate);
#else
    UA_ByteString_copy(&localCertificate, &policy->localCertificate);
#endif
`
In my opinion if in case of default the loadCertificate should not be called.

The memory leak could be due to something with openssl but I think in this case could be avoided.

## Checklist

Please provide the following information:

 - [X] open62541 Version 1.2
 - [X] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Win 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2022-12-12:
@gha88 Thanks for your contribution and sorry for the late reply.
I couldn't reproduce the issue. Could you please provide more information.
Is the memory leak also present in newer open62541 versions? (v1.3 or master)?

NoelGraf, 2023-07-25:
Close the issue for now, reopen it if necessary.

closed 2023-07-25 by NoelGraf