open62541/open62541 #5073: UaExpert 1.6.2 certificate verification error

tboeckel, 2022-04-22:
Since upgrading to open62541 release 1.3 I constantly get certificate verification errors from UaExpert, namely BadInvalidArgument errors, but unfortunately without any further information.
All my open62541 based servers are started with the default settings and do not use any X509 certificates. But nevertheless UaExpert puts up that error window approx every two minutes. I can temporarily accept the certificate but after another few minutes the error window appears again.

However, if I switch back to UaExpert 1.6.1 (or even older UaExpert versions) I don't get such errors. I admit that this might point to a bug in UaExpert 1.6.2 instead of open62541 1.3, but perhaps UaExpert just enhanced its certificate verification and hence exhibits a bug in open62541.

## Used CMake options:
UA_ENABLE_AMALGAMATION=ON
UA_MULTITHREADING=100

## Checklist
 - [X] open62541 Version (release number or git tag): 1.3
 - [X] Other OPC UA SDKs used (client or server): UaExpert 1.6.2
 - [X] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-04-22:
Hmm. Even if that is on the side of UAExpert we should look into that.
I am sure they are testing against a variety of servers and the other probably work fine.

Since you already have a working setup to reproduce, could you make UAExpert write out a detailed trace?
These are the relevant settings to look at:
[image]

mlang-de, 2022-04-22:
If you use SecuritySettings None/None, then this is the desired behaviour from the UaExpert.
https://forum.unified-automation.com/viewtopic.php?f=22&t=7932

tboeckel, 2022-04-25:
So, there is nothing wrong in open62541 and this is all about an attempt of UaExpert to enforce encryption. I will revert to UaExpert 1.6.1 for the time being.

Unfortunately it is still a PITA to get encryption and certificates working in both client and server. Even the supplied examples won't work properly. At least I cannot get the client_encryption/server_encryption combo to work. The client always reports a BadCertificateUriInvalid error, no matter how I generate the client certificate/key combination using the create_self-signed.py script.

closed 2022-04-25 by tboeckel

GoetzGoerisch, 2022-04-25:
@tboeckel please have a look here: https://github.com/umati/Sample-Server