open62541/open62541 #5075: Ongoing Work, Feature Wishlist and Prioritization

jpfr, 2022-04-22:
This meta-issue collects ongoing work to keep on overview on the many work items.
Please add your comment and detailed discussions below in the thread.
This first thread-entry is updated to reflect the current state of the ongoing work.

# Ongoing Work
- [x] Event Filter – Extended Operators
- [ ] EventLoop
  - [x] Integrate the EventLoop into Client and Server
  - [x] Switch Server TCP SecureChannel to the EventLoop
  - [ ] Switch Client TCP SecureChannel to the EventLoop
  - [ ] Switch PubSub Communication to the EventLoop
  - [ ] Switch Discovery (DNS Multicast) to the EventLoop
    - Muddasir from Linutronix is interested
- [ ] PubSub-Publishing of Events
  - PR #4220 was abandonded, needs a final push
- [ ] #4165 Localization Support in the server
  - PR was updated
  - Check with BasysKom
- [ ] Asynchronous Reads/Writes
  - Analogous to async method execution
  - Use case: Fieldbus backend with a delay
- [ ] Full publisher Id support (Martina, KEBA)
  - #4268 Ready for Review
- [ ] Client-Server Security with TPM
  - Store keys in the TPM
  - Also do encryption inside the TPM?
  - Kalycito has started to work on it, but the status is unknown
- [ ] #3981 Publisher Offset PR (Kalycito)
  - Needs refactoring for the master branch
- [ ] PubSub MQTT Subscriber (Kalycito)
  - Base this on the EventLoop control flow
  - #3166 WIP 
- [ ] Parsing of configuration from text-file
  - #4244 JSON5 -> Refactor the default JSON parser
  - Same JSON decoder for config format and the normal JSON parser
- [ ] #3902: Alarms and Conditions for the Client (Kalycito)
  - Appears to be unmaintained
- [ ] #3903: LimitAlarmType and CertificateExpirationAlarmType (Kalycito)
  - Appears to be unmaintained

# PR Housekeeping
- [ ] #4603 MonitoredItems for Nodes that are sometimes hidden
  - Make the behavior configurable in the server config
  - Preparation for review
- [ ] #3975: Fix problems with the old Achilles box
  - The new Achilles box has this fixed on their end
- [ ] #4882 Nodes missing in the reduced information model to unlock additional companion specs for the reduced information model

# Major open issues:
- [ ] #4493 Missing supports for multiple DataSetReaders in the freeze functionality
  - Bilateral clarification Fraunhofer-Kalycito
- [ ] #4858 Multiple Groups + FastPath
- [ ] PubSub option for maximum string length
  - If the length is fixed we need to fill up with zero-bytes
- [ ] PubSub Fastpath multiple messages in one frame
- [ ] open62541.c/.h pair is currently not visible on the hp
  - Resolved with the new homepage

# Feature Wishlist
- ROM-based Binary Nodestore
  - Reduce RAM requirements
  - First implementation in #3411
- Unit tests on Github Actions that builds and runs some of the important companion specs
  - Contingent on the Nodeset Compiler support v1.05
- Diagnostics and auditing
  - [x] Diagnostics Object in the Server
  - Diagnostic/Auditing metadata returned for the services
    - Used by the IOLink CS (optional)
    - Used by the Global Discovery Server
  - PubSub diagnostics
- Full Web Sockets support #4297
  - Feature is still experimental
  - Improve with Event-Loop based networking
- Reduce memory consumption and copying for chunking
  - Currently, enough space for the entire message is allocated
  - Use mmap / MMU to “virtually assemble” the message
- The bsd file format for the DataType definitions is obsolete
  - New xsd file format
  - UAModeler exports both formats
- Better documentation / best practices for PubSub
  - Receiving on windows, etc.
  - Move the PubSub docs to a high-level chapter
- NodeSet Compiler
  - Whitelist node selection for the Nodeset Compiler
    - Backwards chaining to include all dependencies
    - Use this to create the ReducedNodeSet2.xml from a whitelist
- Implement outstanding services
  - Cancel
    - Especially for the Async Method Feature
  - Query
    - SQL-like language for query expressions
- Storing Historical Timeseries on disk
  - Can fork / extend the backend for historical data by BasysKom
- Move multicast DNS to the actively maintained upstream
- Move MQTT-C library to the actively maintained upstream

andreasebner, 2022-11-30:
Please add the "PubSub-Configuration Batch update". This means we offer one information model method which accepts the PubSubConfigurationType  (9.1.3.7). This type contains the whole PubSub-config which can overwrite (simpler) or update (complex) the current configuration.

basyskom-meerkoetter, 2022-12-02:
@jpfr & @andreasebner the "[REVIEW] Locale ids support in the server #4165 Localization Support in the server" can be marked checked. It has been merged.

matkonnerth, 2023-05-29:
I am interested in the query service, here is some work I've done with cypher query language and opcua: https://github.com/matkonnerth/query4Opc

ccvca, 2023-08-25:
@jpfr Can you please add https://github.com/open62541/open62541/issues/5956 to the "Todos for v1.4", as the cmake-installed version of the library is broken at the moment. I've also noted two possible fixes in the issue, but I don't known which of them is according to the ideas of https://github.com/open62541/open62541/commit/7547a8a9dc0b64eaf7a568305d82a78da6a6a6a1#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aL25

OrmEmba, 2024-06-10:
Hello and thanks for all the awesome work you are doing. 
I wanted to ask if there is any progress on porting v1.4 to freeRtos? I am currently working on creating a TSN capable server on an fpga and got v1.3 to run. But after comparing the performance between 1.3 and 1.4 on Linux it is clear, that, in order to reach the targeted responsiveness, we need to change to v1.4. 
If there are  already some starting points i would be happy to contribute.

m-obi, 2024-11-19:
Can we add autoinject_library and the generated header-files from src_generated to the install/export command?

xR3b0rn, 2026-08-20:
@jpfr what is missing for RBAC to be production ready?