open62541/open62541 #5078: A client cannot write a value to a node after subscribing to it

nihao-git, 2022-04-26:
## Description

A client cannot write a value to a node after subscribing to it

Call write value API:UA_Client_writeValueAttribute

Error after calling write value API after subscribing node:
[2022-04-26 17:58:22.240 (UTC+0800)] warn/channel       Connection 436 | SecureChannel 3208649311 | Receiving the response failed with StatusCode BadConnectionClosed
[2022-04-26 17:58:22.240 (UTC+0800)] warn/channel       Connection 436 | SecureChannel 3208649311 | Receiving the response failed with StatusCode BadConnectionClosed
[2022-04-26 17:58:22.240 (UTC+0800)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2022-04-26 17:58:22.241 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
[2022-04-26 17:58:22.243 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

UBehrens, 2022-07-19:
Good morning everybody!

Has any news shown up in the context of this thread? I have recently faced some similar issues however not with exactly reproducable behaviour. Even when having two machines with same hardware and same software installations, one of them shows above behaviour from time to time, while the other one seems to be resistent.

Both are using open62541 1.2.2 on Ubuntu 20.04 on the client side and two CODESYS servers with identical hardware for which I can not provide detailed information. But since the the problem does always show up on the same client, even when I exchange the server/client connections, I assume it might be a generic bug or a configuration setup problem on the client side.

I would be helpful for any assistance with this. Thanks a lot, Uwe.

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier