open62541/open62541 #5079: The Circular Historian returns all of the saved history instead of the subset requested.

YperionBristogiannis, 2022-04-26:
## Description

The Circular Historian (#2835, #4740) returns all of the saved history instead of the subset requested. For example, if I have a history of 10 different values and request the history of a timespan that should include only the last 8 values, I will still get all 10 values.

Some context: Even though I am using `UA_HISTORIZINGUPDATESTRATEGY_VALUESET`, my updates come at fixed intervals so I can predict how many values I should get within a requested timespan depending on the node (some nodes are updated on different intervals).

This is a problem because I get the whole history for every subscribed node even though it is not required or wanted. In addition, I do not have control over the history timespan that will be requested apart from a general limit of "at least *n* minutes with a max of *m* minutes" thus setting separate historians with different timespans is not going to help.

Currently, this is forcing me to use the non-circular historian that indefinitely increases in memory usage which then forces me to restart the server app thus losing the recently obtained history that I need to keep.

## Background Information / Reproduction Steps

This can be reproduced by altering the unit test `Server_HistorizingStrategyValueSet` in the `check_server_historical_data_circular.c` file:

I change the starting time of the requested history by replacing the lines 229 to 235
```c
    UA_DateTime start = UA_DateTime_now();
    UA_fakeSleep(100);
    for (UA_UInt32 i = 0; i < 15; ++i) {
        retval = setUInt32(client, outNodeId, i);
        ck_assert_str_eq(UA_StatusCode_name(retval), UA_StatusCode_name(UA_STATUSCODE_GOOD));
        UA_fakeSleep(100);
    }
```
with the following:
```c
    for (UA_UInt32 i = 0; i < 7; ++i) {
        retval = setUInt32(client, outNodeId, i);
        ck_assert_str_eq(UA_StatusCode_name(retval), UA_StatusCode_name(UA_STATUSCODE_GOOD));
        UA_fakeSleep(100);
    }

    UA_DateTime start = UA_DateTime_now();

    for (UA_UInt32 i = 7; i < 15; ++i) {
        retval = setUInt32(client, outNodeId, i);
        ck_assert_str_eq(UA_StatusCode_name(retval), UA_StatusCode_name(UA_STATUSCODE_GOOD));
        UA_fakeSleep(100);
    }
```
This will cause the unit test to fail at
```c
ck_assert(data->dataValues[j].sourceTimestamp >= start && data->dataValues[j].sourceTimestamp < end);
```
which implies that the starting date is wrong.

If we comment-out this assert command then the test passes even though the check is for the whole history and I only requested a subset of it thus ideally it would have failed. Adding print statements helps see this.

We can apply the same on the non-circular historian in the `check_server_historical_data.c` and see that the assert command that fails for the circular historian passes for the non-circular but then the check fails when it checks the values in the history. This makes sense because the test expects the whole history. Therefore, the issue is *only* for the *circular* historian.

Used CMake options:
```bash
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): *master branch*
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: *Ubuntu 20.04*
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached: *not self-contained code but described required changes to existing code*
 - [x] Critical issue: *It is critical for me for the reasons mentioned in the desciption.*

jpfr, 2022-05-07:
@luibass92 can you have a look at this?

luibass92, 2022-06-06:
Hello @YperionBristogiannis and thanks for your feedback.

The circular implementation is intended to solve the problem related to memory leak and it does not handle any features related to the timestamp. Check the function `serverSetHistoryData_backend_memory_Circular` in `plugins/historydata/ua_history_data_backend_memory.c`.
Moreover the circular buffer is not ordered by timestamp.

Of course it could be improved adding also timestamp features but in that case you should chose a logic to discard old values in the circular buffer. E.g. Would you discard the oldest inserted value or the value with the oldest timestamp ?

As of now the buffer will keep simply the latest value that arrives (without checking if its timestamp is the most recent or not) and if the buffer is full it will remove the "oldest" saved value, even though this value has a timestamp that it is actually more recent.

This choice has been done just for simplicity and to keep the insertion fast (avoiding to binary search the entire buffer at each insertion).

YperionBristogiannis, 2022-06-08:
Hi @luibass92, thank you for looking into this.

I work under the assumption that the values are sent to the server in a timely fashion. So the order in which they are *inserted* in the historian (circular or not) also implies that they are newer than their previous one. In other words, I don't think that anything should change when it comes to how the values are saved in the circular buffer.

What I tried to show in the issue description is that if I request a subset of the history, I still get all of the history. As an example, if I have inserted **6** values (A,B,C,D,E,F) in a circular buffer that fits **4** values, I understand that it will be "saved" as: `E,F,C,D`. If I request a history span that covers the **last 3** values, I would expect to receive `D,E,F`. Instead, I receive **all 4** values in the history. You can see this in action as I describe in the "reproduction steps" in the issue description.

I think this could be addressed when handling a request for a subset of the history (I say "I think" because I am not aware of how this works internally). So when such a request occurs, it is decided on the spot which elements are to be sent (perhaps using the binary search you mentioned). Do you think this makes sense?

luibass92, 2022-06-12:
@YperionBristogiannis yes I understand your point of view.
It definitely makes sense for your use case, but this would require to re-order the elements of the buffer at each insertion.

See the code of the "normal" implementation of `serverSetHistoryData_backend_memory()`.

Expecially the part

```
UA_DataValue_copy(value, &newItem->value);
    size_t index = getDateTimeMatch_backend_memory(server,
                                                   context,
                                                   NULL,
                                                   NULL,
                                                   nodeId,
                                                   timestamp,
                                                   MATCH_EQUAL_OR_AFTER);
    if (item->storeEnd > 0 && index < item->storeEnd) {
        memmove(&item->dataStore[index+1], &item->dataStore[index], sizeof(UA_DataValueMemoryStoreItem*) * (item->storeEnd - index));
    }
```

As you can see at each insertion an index is calculated to understand where the element should be inserted considering its timestamp and, if needed, a memmove is performed to put the data in the correct place. This will ensure that the array is always ordered by timestamps and so the get method that you are complaining about will work properly.

Now consider an use cause with a circular buffer of size 5 for example.
After you insert the 5th element, any subsequent element would be inserted in a similar way keeping the array always ordered by timestamp. Cause of the circular buffer structure, each insertion starting from the 6th will cause a call to `memmove`.
This is really expensive in terms of computational power and I am not sure what this would cause in terms of performances. For sure it is not a very good solution for all the use cases which require high frequency insertions.

**E.g.**

Let's suppose we have an array of size 5 and that we already inserted 5 elements (in a timely fashion):

`A, B, C, D, E`

Now the 6th element arrives, we have to basically remove A sliding the entire array to the left and insert the new element at the end

`A,B,C,D,E` ----memmove---> `B,C,D,E,_` ----insertion----> `B,C,D,E,F`

The process will repeat for the subsequent elements `G` that requires the removal of `B` and so on so forth.

Hope my point of view is clear.