open62541/open62541 #5080: Bug: Can no longer set server port when using security policy

rexsum, 2022-04-26:
## Description

Recent commits starting with `26b86ed` on April 20th have broken my server. It's very difficult to find the offending code and root cause the problem because that commit is quite extensive. The most recent master commit also don't work for me.

It seems that I can no longer set the port when configuring my server with security policy. It ALWAYS defaults to port 4840 rather than using the port I explicitly specify.

## Background Information / Reproduction Steps

This is the command in my application that no longer works as intended:

``` 
UA_ServerConfig_setDefaultWithSecurityPolicies(*server, port, &certificate, &private_key, trustlist, trustlist_size, NULL, 0, NULL, 0);
```

I hope that this is sufficient information to resolve this issue. If not, I can try to pull together a small application to demonstrate the problem if necessary, but it would take time to create that.

I need this bug fixed because I have multiple servers and it's important that I be able to set the port for each.

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo \
      -DBUILD_SHARED_LIBS=ON \
      -DUA_LOGLEVEL=300 \
      -DUA_MULTITHREADING=100 \
      -DUA_FORCE_WERROR=OFF \
      -DUA_NAMESPACE_ZERO=REDUCED \
      -DUA_ENABLE_PUBSUB=ON \
      -DUA_ENABLE_PUBSUB_DELTAFRAMES=ON \
      -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON \
      -DUA_ENABLE_PUBSUB_ETH_UADP=ON \
      -DUA_ENABLE_PUBSUB_ENCRYPTION=ON \
      -DUA_ENABLE_PUBSUB_BUFMALLOC=ON \
      -DUA_ENABLE_IMMUTABLE_NODES=ON \
      -DUA_ENABLE_MALLOC_SINGLETON=ON \
      -DUA_ENABLE_ENCRYPTION_TPM2=ON \
      -DUA_ENABLE_ENCRYPTION=OPENSSL \
      -DUA_ENABLE_METHODCALLS=ON \
      -DUA_ENABLE_SUBSCRIPTIONS=ON \
      -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON \
      -DUA_ENABLE_HISTORIZING=ON \
      ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

rexsum, 2022-04-29:
Ok, so how do I fix my code to set the port? Even the code example still uses UA_ServerConfig_setDefaultWithSecurityPolicies to set the port so I have no idea how to proceed.

rexsum, 2022-04-29:
Thanks to @lightmare for pointing me in the right direction. 

I'm posting this update for everyone else's benefit who may be wondering how to fix their code. 

After wasting several hours of my time reverse engineering the library AGAIN due to backward compatibility was once again having been broken, with NO change log, NO documentation, NO updates to the example code, NO comments in the code regarding the redesign, and NO output to the log indicating that the old method of port handling has been deprecated.

Here's what you need to know:
- Setting server port is no longer done during configuration
- Setting the server port is now done when starting up the server

Ex.
```
    UA_String* urls = (UA_String*)UA_Array_new(1, &UA_TYPES[UA_TYPES_STRING]);
    urls[0] = UA_STRING_ALLOC("opc.tcp://:4841");
    server->config.serverUrlsSize = 1;
    server->config.serverUrls = urls;
    UA_Server_run_startup(server);
```

Hope this helps

jpfr, 2022-05-02:
Hey @rexsum,

These are all good points.
We will add changelog and examples for the next release.

I cannot agree that there is no documentation.
Because the documentation is generated from the header files where the new configuration format is explained.

One important point is that all configurations remain possible in principle.
So if you have a scenario that is no longer feasible, please say so and we will add the support.

jpfr, 2022-05-02:
Right, the portnumber in UA_ServerConfig_setDefaultWithSecurityPolicies has to be evaluated.
That’s a bug / oversight.

jpfr, 2022-05-02:
The motivation for the change:
We need a unified way to support multiple protocols / ports and to also specify hostnames to listen on.
For example a server might listen on 4840 with a public hostname with only encrypted endpoints.
And to allow unencrypted traffic only from an internal network on another port.

rexsum, 2022-05-02:
Thanks for the reply @jpfr 

Change isn't the issue, the library will always continue to evolve, that's expected and not at all a problem. Even bugs occasionally slipping through is not unexpected.

What frustrates me, as a long time user of this library, is when changes are made that breaks backward compatibility w/o sufficient heads up. That results in a huge time-sink for me.

I'm glad that you've agreed to providing a changlog in the future. That one thing alone will help alleviate this major pain point. Updates to the user guide and keeping the example code in sync with the API will also be a tremendous help, thank you.

Let me know when a decision has been reached on the UA_ServerConfig_setDefaultWithSecurityPolicies() function so I can update my code accordingly.

jpfr, 2022-05-05:
Solved via #5105.

rexsum, 2022-05-10:
@jpfr I finally had some time to test your PR and found that while technically it does fix the issue with UA_ServerConfig_setDefaultWithSecurityPolicies() not respecting the port number, I still however have issues..........

I find that my other application which uses UA_ServerConfig_setMinimal() also has the same problem. It ignores the port number just as UA_ServerConfig_setDefaultWithSecurityPolicies() was doing.

In addition to that, I found a new problem that shutdown of the server when using PubSub always hangs. It is now necessary to kill the process to stop the application. Here's a brief log output that shows the server as it shuts down and where it hangs:

```
[2022-05-10 15:05:43.523 (UTC-0700)] info/eventloop	Stopping the EventLoop
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	UDP	| Shutting down the ConnectionManager
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	TCP	| Shutting down the ConnectionManager
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	TCP 6	| Cannot close - already closing
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	TCP 5	| Cannot close - already closing
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	TCP 7	| Socket closed
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	TCP 6	| Socket closed
[2022-05-10 15:05:43.523 (UTC-0700)] info/network	TCP 5	| Socket closed
[2022-05-10 15:05:43.525 (UTC-0700)] info/eventloop	The EventLoop has stopped
[2022-05-10 15:05:43.525 (UTC-0700)] info/server	PubSub cleanup was called.
```

Both of these problems can be demonstrated with the PubSub examples. I'm specifically testing with `pubsub_publish_encrypted.c` but have found problems with others as well.

I can file new issues for these if necessary. Let me know if you prefer that to extending this issue like I've done here. As it stands now however, it is still not possible for me to use any newer commits.

closed 2022-05-23 by rexsum