open62541/open62541 #5082: 1.3 release missing official docker image

lordgamez, 2022-04-27:
## Description

Currently there are only two images uploaded in the official docker repository on Dockerhub: 1.0 (latest tag set to this) and a master image tag last updated 9 months ago. It would be great to pull the docker images of major releases, but at least the last released 1.3 version from the official Dockerhub repository. Would it be possible to upload it there?

powersj, 2022-10-26:
Hi,

We are using the open62541 dockerhub image in integration tests for Telegraf. It would be great to see the dockerhub image updated as requested here.

Thanks!

LarsStegman, 2022-11-23:
@jpfr I am using the image in some of my tests but they are failing because the certificate has expired almost a year ago. Is it possible to release a new version?

GoetzGoerisch, 2022-11-23:
@jpfr currently the container image is not build automatically with a release. As hosting the container image is now easy at ghcr.io instead of docker hub, this might be easy to integrate in the CI?

@LarsStegman  and @powersj I don't know your correct need, we are building a sample server with a few companion specification instances here: https://github.com/umati/Sample-Server.