open62541/open62541 #5085: Server Subscriptions/Monitored Items/DataChange Notifications issue

serega1103, 2022-04-28:
## Description
Hi!

I am a beginner and just starting to master OPC UA and open62541.

I have an absolutely trivial task. I am writing a server and I want clients to be able to track the change in the value of my elements.

Please help me find the answers to these questions:
1. Do I understand correctly that the open62541 library has a built-in mechanism for tracking changes in the values of MonitoredItems on the server side?
2. If I want to achieve the easiest way so that clients can receive notifications about variable changes, do I not need to write any additional infrastructure on the server for this, because it is already built "under the bonnet"?
3. Let's assume that the code of my server is below. Do I understand correctly that this code is enough so that the client can already track the change in the variable value using retval = UA_Client_Subscriptions_addMonitoredItem(client, subId, UA_NODEID_STRING(1, "the.answer"), UA_ATTRIBUTEID_VALUE, handler_TheAnswerChanged, NULL, &monId, 5000), and the server "under the bonnet" will track itself changing the value of the myIntegerNodeId variable and I don't need to write any additional code for this on the server?

```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static void
addVariable(UA_Server *server) {
    /* Define the attribute of the myInteger variable node */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 myInteger = 42;
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    attr.description = UA_LOCALIZEDTEXT("en-US","the answer");
    attr.displayName = UA_LOCALIZEDTEXT("en-US","the answer");
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Add the variable node to the information model */
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
}

static volatile UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
    UA_ServerConfig* config = UA_Server_getConfig(server);
    config->verifyRequestTimestamp = UA_RULEHANDLING_ACCEPT;
}
```

I'm sorry, but I couldn't find explicit answers to these questions.
It seems to me that the understanding of this is very lacking for beginners.
Thank you in advance for your help!
P. S. Thanks to all contributors for a great library!

## Background Information / Reproduction Steps
Used CMake options:
```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON -G "Visual Studio 17 2022" -A Win32 -S .. -B "build32"
```

## Checklist
 - [x] open62541 Version: v1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

neojaw, 2022-04-29:
Hello,

yes, the server in standard mode already provides the necessary functionality so that a client can subscribe to it. No additional work or configuration needed. Your server code seems incomplete though, I don't see a place where you actually start it.

Talking about the client side, the function `UA_Client_Subscriptions_addMonitoredItem()` you mentioned does not exist anymore, it was removed in version 0.4. You have to create a subscription with `UA_Client_Subscriptions_create()` which gives you, among other things, a subscription id. Then you can create monitored items under that id with `UA_Client_MonitoredItems_createDataChange()`, supplying a callback function.

See the documentation for subscriptions: https://open62541.org/doc/1.2/client_subscriptions.html

serega1103, 2022-04-29:
@neojaw, thank you very much for the answer!
Yes, I forgot to add the place where the server starts.

The corrected code should look like this:
```
int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig* config = UA_Server_getConfig(server);
    UA_ServerConfig_setDefault(config);    
    config->verifyRequestTimestamp = UA_RULEHANDLING_ACCEPT;

    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

Your answer explained everything to me, thank you!
I close the question.

closed 2022-04-29 by serega1103