open62541/open62541 #5101: Question: Browsing nodes with BaseDataVariableType-reference

chs-1980, 2022-05-05:
Hi everybody,

This is more a usage question than a bugreport but i am at a loss currently and am stuck at the moment.

I implemented a OPC-UA-plugin (we support both client and server-mode) for our application and now we have a customer which wants to use the plugin to connect to a Siemens PLC. I can connect to the OPC UA-server and i am able to browse but the problem is the variables which would the interesting ones are not found during browsing. The interesting thing is that the NodeClass of the "problematic" vars is Variable so i should be able to find them because browsing other vars with the same NodeClass works without any problems. So i checked what could be the difference between one variable that is found and one which is not. And the only difference i can see is that they have different HasTypeDefinition-References: The one that works is a property (PropertyType-ref) and the other has a BaseDataVariableType-reference. Currently i am configuring my browserequest like this:

`	UA_BrowseRequest bReq;
	UA_BrowseRequest_init(&bReq);
	bReq.requestedMaxReferencesPerNode = 0;
	bReq.nodesToBrowse = UA_BrowseDescription_new();
	bReq.nodesToBrowseSize = 1;
	bReq.nodesToBrowse[0].browseDirection = UA_BROWSEDIRECTION_FORWARD;
	bReq.nodesToBrowse[0].nodeClassMask = 3;
	bReq.nodesToBrowse[0].nodeId = nodeId; /* browse objects folder */
	bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; /* return everything */
	UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);`

I assume my problem is the nodeClassMask but no matter what i set this to (or even remove the line completely) the BaseDataVariableType-variables are ignored. 

Here is a screenshot from UAExpert where i show the attributes and references of two nodes where one works (Icon) and the other doesn't (EingangReal)

[image]

Do you have an idea what I am missing?

Thanks,

Christoph

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3
 - [x] Other OPC UA SDKs used (client or server): UAExpert 1.5.1
 - [x] Operating system: Windows 10 21H1
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-05-05:
Which reference Typ is used to point to these variables?

chs-1980, 2022-05-06:
The node class of the Inputs node is Object and the HasTypeDefinition-reference (i assume that it's what you are interested in) is FunctionalGroupType.

Here is a screenshot of UAExpert where i selected the "problematic" node and you can probably see all relevant information:

[image]

chs-1980, 2022-09-16:
Hi,

Now i have a second customer asking the same question so i have to bump this again... I would be thankful for any idea, anything i could try/test in order to get this to work...

shaohuazhang1, 2022-10-11:
Hi,
in your screenshort,the node you can browsed NodeId(3,5204),namespaceIndex is 3,but namespaceIndex is 0 in browsename(0,"icon").In another node the namespaceIndex is the same,3.
Have you solved this problem?

UlfJohnsson, 2022-11-11:
So if anyone still wanna know I solved this with my Siemens PLC.
What you gotta do is perform a UA_BrowseNextRequest. You'll probably find that your UA_BrowseRequest contains a "continuationpoint" and if it does use that in the next request and you'll get what you seek.
Something like this:

```
if(result.continuationPoint.length != 0) {
  UA_BrowseNextRequest nreq;
  UA_BrowseNextRequest_init(&nreq);
  nreq.continuationPointsSize = 1;
  nreq.continuationPoints = UA_ByteString_new();
  UA_ByteString_copy(&result.continuationPoint, nreq.continuationPoints);
  nreq.releaseContinuationPoints = UA_FALSE;

  UA_BrowseNextResponse nresponse = UA_Client_Service_browseNext(_d->client, nreq);
  if(nresponse.resultsSize == 1 && nresponse.results)
    UA_BrowseResult *result = nresponse.results; // will contain missing nodes
}
```

chs-1980, 2022-11-11:
@UlfJohnsson Very interesting! Thanks for the heads up. As usual the problem is that it was absolutely urgent a few weeks back and now the same customer said (because i found out that i for some reason unknown to me i simply ignored variables with numeric nodeids altogether and i fixed that and my guess is that this was the main problem) that right now this is not so important any more because it is almost impossible to get Siemens PLCs anyway ATM. But i'll integrate that sometime in the future and i'll close this issue for the moment!

@all Thanks for the ideas

closed 2022-11-11 by chs-1980