open62541/open62541 #5102: Valgrind reports errors on Raspberry Pi 3 device when running server examples

jonasbindrim, 2022-05-05:
## Description

After using this library on my Ubuntu-PC for quite some time, I`m trying to use this library on a Raspberry Pi 3. After successfully compiling and running the tutorial_server_firststeps example on my pi, I checked whether valgrind would report any errors.  While checking with valgrind I was receiving many different errors and at some point valgrind would freeze.

Note: When compiling and running on my Ubuntu-PC it works totally fine and valgrind does not report any issues.

## Valgrind logs

```bash
==7205== Memcheck, a memory error detector
==7205== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
==7205== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==7205== Command: ./tutorial_server_firststeps
==7205==
--7205-- WARNING: Serious error when reading debug info
--7205-- When reading debug info from /usr/lib/arm-linux-gnueabihf/ld-2.28.so:
--7205-- Ignoring non-Dwarf2/3/4 block in .debug_info
--7205-- WARNING: Serious error when reading debug info
--7205-- When reading debug info from /usr/lib/arm-linux-gnueabihf/ld-2.28.so:
--7205-- Last block truncated in .debug_info; ignoring
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401A5B0: index (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401A5B4: index (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x4007F0C: _dl_dst_count (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x4008154: expand_dynamic_string_token (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401AA60: strlen (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401AA64: strlen (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x4017F90: malloc (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x4017F9C: malloc (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B5C8: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B5E8: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B5F8: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B614: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B61C: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B644: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Use of uninitialised value of size 4
==7205==    at 0x401B644: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Use of uninitialised value of size 4
==7205==    at 0x401B66C: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
==7205==
==7205== Conditional jump or move depends on uninitialised value(s)
==7205==    at 0x401B680: memcpy (in /usr/lib/arm-linux-gnueabihf/ld-2.28.so)
[... 216 lines cut ...]

Reproduction Steps:

1.  Clone the repository
2. Switch to branch 1.3
3. Build the library and the examples
4. Execute the tutorial_server_firststeps with valgrind (`valgrind ./tutorial_server_firststeps`)
5. Receive tons of errors

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=ON ..
cmake --build .
```

## Checklist

 - [x] open62541 Version (release number or git tag): Branch 1.3; Commit: 41de7e470c19bdeb4f404dd5e96e808db488ef8c
 - [ ] Other OPC UA SDKs used (client or server): None
 - [x] Operating system: Raspbian 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-05-05:
Hmm. Can you try and fix the debug info?
The log Output is very limited without it.

jonasbindrim, 2022-05-06:
Thanks for the quick reply. I fixed the debug errors by manually updating valgrind to a newer version (3.19).
The new valgrind looks like this: [valgrind.log](https://github.com/open62541/open62541/files/8638436/valgrind.log)

jpfr, 2022-05-07:
There is a memory problem indicated in UA_DateTime_now.
That function is very small and looks clean.
I suppose that most or all of these valgrind issues are false positives.

closed 2022-05-17 by jonasbindrim