open62541/open62541 #5112: Build a simple server on EK-TM4C1294XL target

PavanJoshiS, 2022-05-07:
## Description

I am having trouble building a simple open62541 server on EK-TM4C1294XL.I followed the steps mentioned in the following page
https://open62541.org/doc/current/building.html. But I am getting a lot of errors. Is anyone successful building the open62541 on TIVA-c boards.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_ARCHITECTURE=freertosLWIP -DUA_ENABLE_AMALGAMATION=ON ../
```

jpfr, 2022-05-09:
Try the 1.3 branch.
The master branch is undergoing refactoring and freertos has not yet kept up.
Alas, we don’t have continuous integration for freertos.
So it needs some manual fixing up before releases.

PavanJoshiS, 2022-05-27:
Thanks for the messages. The TI boards come with their own compiler and the compiler is missing os libraries like wsock.lib and ws2_32.lib. I could build the applications in visual studio without any errors by linking the mentioned libraries. How to proceed in this case? Is anyone able to build an application using TI compiler?