open62541/open62541 #5113: Deprecated SSL version in MQTT TLS implementation

ivan-ushakov, 2022-05-08:
## Description
Current MQTT TLS implementation uses old SSL 2.3 version and should be updated to TLS 1.2 or TLS 1.2.

Here is how OpenSSL context is created:
```
// OpenSSL 1.0 doesn't have TLS_client_method(), use SSLv23 and forbid SSLv2 and SSLv3
SSL_CTX *ctx = SSL_CTX_new(SSLv23_client_method());
SSL_CTX_set_options(ctx, SSL_OP_NO_SSLv2 | SSL_OP_NO_SSLv3);
```

OpenSSL provides `TLSv1_2_client_method(void)` and also we could set version directly with `SSL_CTX_set_min_proto_version` and `SSL_CTX_set_max_proto_version`.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch
 - [x] Critical issue

ivan-ushakov, 2022-05-08:
This also leads to problem with `mosquitto` MQTT server:
```
1652031469: New connection from ?:? on port 1883.
1652031469: OpenSSL Error[0]: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
1652031469: Client <unknown> disconnected: Protocol error.
```

jpfr, 2022-05-09:
This is generally fixed in #5086 .

NoelGraf, 2022-05-10:
Do not be confused by the name (it does NOT mean that only SSLv2 or SSLv3 will be used). The name is like that for historical reasons. Normally you should always use SSLv23_method to get the latest available TLS version.

ivan-ushakov, 2022-05-10:
> Do not be confused by the name (it does NOT mean that only SSLv2 or SSLv3 will be used). The name is like that for historical reasons. Normally you should always use SSLv23_method to get the latest available TLS version.

Yes, they have this in source code:
```
#define SSLv23_client_method    TLS_client_method
```
but I guess this was done for the legacy code. In new code we should use `TLS_client_method` directly.

closed 2022-05-10 by ivan-ushakov