open62541/open62541 #5132: Changing hostname

strager14, 2022-05-18:
## Description
I want to change the server url to a custom one, like "opc.tcp://MyServer:4840". How can i do this? 
Function "UA_ServerConfig_set_customHostname" was removed in some version.
I tried the following code, it doesn't work
## Background Information / Reproduction Steps
```
UA_String uastrValue = UA_STRING("opc.tcp://MyServer:4840");
UA_String_deleteMembers(config->serverUrls);
UA_String_copy(&uastrValue, config->serverUrls);
```
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached

Muddasir-Shakil, 2022-05-18:
Hi @strager14 ,
Try this:
 UA_String customHost = UA_STRING("localhost");
UA_String_clear(&config->customHostname); 
UA_String_copy(&customHost, &config->customHostname); 
BR

strager14, 2022-05-18:
> Привет@strager14, Попробуйте следующее: UA_String customHost = UA_STRING("localhost"); UA_String_clear(&config->customHostname); UA_String_copy(&customHost, &config->customHostname); BR

Unfortunately I incorrectly specified the branch used. Branch master, not 1.3. There is no customHostname in the master version

jpfr, 2022-05-23:
You found the correct variable.
But you are not using it the correct way.

`config->serverUrls` is an array of url strings, not a single url string.
See the documentation generated from /include/open62541/server.h

See `plugins/ua_config_default.c` for examples how this is set.
Don't forget to set `config->serverUrlsSize` as well.

closed 2022-05-23 by jpfr

neojaw, 2022-06-02:
@StenGruener What do you mean?

StenGruener, 2022-06-02:
@neojaw sorry, did not notice it was closed