open62541/open62541 #5142: Bug: Stackoverflow for value callback when using Prosys OPC UA Browser

Rehcsif, 2022-05-24:
## Description
When using Prosys OPC UA Browser there is a StackOverflow when monitoring an Object/Node, which is updated via a callback. As far as I have seen OPC UA Browser sets a sampling interval of 0.0 for newly added nodes (the interval can be adjusted, however the stackoverflow already occured).

## Background Information / Reproduction Steps
For a sampling interval of 0.0 open62541 seems to act differently. The stack responds to value changes directly. But this also means, that when a Node is updated in a callback via `UA_Server_writeValue()`, that the callback `beforeReadTime()` is triggerd once again and so on until a stackoverflow occurs.

One can reproduce the issue by installing OPC UA Browser 4.1.2-364 and building and executing the example from http://www.open62541.org/doc/current/tutorial_server_datasource.html

Search for the "value callback"-Node and click on "Monitor data". A subscription with sampling interval 0.0 will be created and a stackoverflow will occur.

Used CMake options:

```bash
cmake  -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_PUBSUB=ON -DUA_BUILD_EXAMPLES=ON
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3
 - [x] Other OPC UA SDKs used (client or server):  OPC UA Prosys Browser 4.1.2-364
 - [x] Operating system: windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

strager14, 2022-05-25:
If i understand you correctly, you can use 
```
if (sessionId->identifier.numeric != 1)
{
...
}
```
in function beforeReadTime(). This will ignore read requests from the server itself.

marwinglaser, 2025-03-18:
Solved

closed 2025-03-18 by marwinglaser

paperino1983, 2025-07-14:
> If i understand you correctly, you can use
> 
> ```
> if (sessionId->identifier.numeric != 1)
> {
> ...
> }
> ```
> 
> in function beforeReadTime(). This will ignore read requests from the server itself.

For info, now this is not anymore true, _sessionId_ are identified via GUID. But I really don't find where and how the _sessionId_: ns:1, g:1000000-0000-0000-0000-000000000000 is created internally