open62541/open62541 #5143: opc server does not start

bhocheLunovu, 2022-05-24:
## Description
When I try to start the simple server from the xamples I get the result 0x80050000
The code is working on my dev-pc but not on my laptop

Does anyone has a hint how to find the problem?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

vvpnet, 2022-05-24:
Running under windows OS? Mingw? Maybe the library dependencies are not specified in env PATH?

bhocheLunovu, 2022-05-24:
running on windows10 with vs2019 on both machines

closed 2022-05-24 by bhocheLunovu