open62541/open62541 #5151: SERVER TUTORIAL NOT WORKING

Fabrizio997, 2022-05-26:
## Description 

As title says I installed the open62541 on ubuntu (running through virtualization on mac), I followed the first simple steps to build and create a simple server but it doesn't run correctly

server is this:

    #include <open62541/plugin/log_stdout.h>
    #include <open62541/server.h>
    #include <open62541/server_config_default.h>
    
    #include <signal.h>
    #include <stdlib.h>
    
    static volatile UA_Boolean running = true;
    static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
    }
    
    int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);
        
    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
    
    UA_StatusCode retval = UA_Server_run(server, &running);
    
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
    } 

and here is the log of when I launch the server 

    fabrizio@fabrizio:~/Desktop/build2/open62541/build$ ./myServer
    [2022-05-26 16:08:37.636 (UTC+0000)] warn/userland	ServerUrls already set. Overriding.    
    [2022-05-26 16:08:37.636 (UTC+0000)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
    [2022-05-26 16:08:37.636 (UTC+0000)] info/server	AccessControl: Anonymous login is enabled
    [2022-05-26 16:08:37.636 (UTC+0000)] info/server	AccessControl: x509 certificate user authentication is enabled
    [2022-05-26 16:08:37.636 (UTC+0000)] warn/server	Username/Password Authentication configured, but no encrypting     SecurityPolicy. This can leak credentials on the network.
    [2022-05-26 16:08:37.636 (UTC+0000)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
    [2022-05-26 16:08:37.636 (UTC+0000)] info/eventloop	Starting the EventLoop
    [2022-05-26 16:08:37.636 (UTC+0000)] info/network	TCP	| Listening on all interfaces
    [2022-05-26 16:08:37.636 (UTC+0000)] info/network	TCP 4	| New server socket for "0.0.0.0" on port 4840
    [2022-05-26 16:08:37.636 (UTC+0000)] info/network	TCP 5	| New server socket for "::" on port 4840
    [2022-05-26 16:08:37.636 (UTC+0000)] warn/network	TCP	| getaddrinfo lookup for "fabrizio" on port 4840 failed (None)

I cannot understand why it gives me  the message [2022-05-26 16:08:37.636 (UTC+0000)] warn/network	TCP	| getaddrinfo lookup for "fabrizio" on port 4840 failed (None) 

## Background Information / Reproduction Steps

Used CMake options:

cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON ..

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): [e84b2d1](https://github.com/open62541/open62541/commit/e84b2d18039b01b2988f8a120f6670db395441e2) commit ec42e5c65f9211eba13a5346822ab017b617bfff
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu 22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

kimim, 2022-05-27:
You may need to check with `ipconfig` whether you configured the IP address.
`getaddrinfo` will look up IP address for your hostname `fabrizio`.

Fabrizio997, 2022-05-27:
> You may need to check with `ipconfig` whether you configured the IP address.
> `getaddrinfo` will look up IP address for your hostname `fabrizio`.

thanks a lot, however using ifconfig the output I get is the seguent

fabrizio@fabrizio:~$ ifconfig
enp0s6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.64.8  netmask 255.255.255.0  broadcast 192.168.64.255
        inet6 fdf0:b83:6fde:5530:c03a:b8ff:fe84:209f  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::c03a:b8ff:fe84:209f  prefixlen 64  scopeid 0x20<link>
        ether c2:3a:b8:84:20:9f  txqueuelen 1000  (Ethernet)
        RX packets 318  bytes 293949 (293.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 250  bytes 29730 (29.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 141  bytes 12561 (12.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 141  bytes 12561 (12.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

jpfr, 2022-05-30:
Fixed via #5158.
Can you verify this solves tour issue?

Fabrizio997, 2022-05-30:
> Fixed via #5158.
> Can you verify this solves your issue?

I get this error while building with your : https://github.com/jpfr/open62541.git

  error: cast between incompatible function types from ‘void (*)(UA_Server *, UA_SecureChannel *, UA_Session *, const UA_ActivateSessionRequest *, UA_ActivateSessionResponse *)’ to ‘void (*)(UA_Server *, UA_Session *, const void *, void *)’ [-Werror=cast-function-type]
  23022 |         *service = (UA_Service)Service_ActivateSession;
        |                    ^
  /home/fabrizio/Desktop/prova1/open62541/build/open62541.c: In function ‘openSecureChannel’:
  /home/fabrizio/Desktop/prova1/open62541/build/open62541.c:32251:24: error: cast between incompatible function types from ‘UA_StatusCode (*)(void *, UA_SecureChannel *, UA_MessageType,  UA_UInt32,  const UA_ByteString *)’ {aka ‘unsigned int (*)(void *, UA_SecureChannel *, UA_MessageType,  unsigned int,  const UA_String *)’} to ‘void (*)(UA_Client *, void *, UA_UInt32,  const void *)’ {aka ‘void (*)(UA_Client *, void *, unsigned int,  const void *)’} [-Werror=cast-function-type]
  32251 |         ac->callback = (UA_ClientAsyncServiceCallback)processDecodedOPNResponse;
        |                        ^

jpfr, 2022-05-30:
This is a completely unrelated error.
Are you on the branch [jpfr:non_numeric_server_hosts](https://github.com/jpfr/open62541/tree/non_numeric_server_hosts)?

Nevermind, I already merged to master.

Fabrizio997, 2022-05-30:
> This is a completely unrelated error. Are you on the branch [jpfr:non_numeric_server_hosts](https://github.com/jpfr/open62541/tree/non_numeric_server_hosts)?
> 
> Nevermind, I already merged to master.

ok, I m trying again with master branch then

Fabrizio997, 2022-05-30:
> This is a completely unrelated error. Are you on the branch [jpfr:non_numeric_server_hosts](https://github.com/jpfr/open62541/tree/non_numeric_server_hosts)?
> 
> Nevermind, I already merged to master.

the past problem with ([2022-05-26 16:08:37.636 (UTC+0000)] warn/network	TCP	| getaddrinfo lookup for "fabrizio" on port 4840 failed (None)) is no more there 
but now the when I launch the server it stuck with:
   [2022-05-30 10:30:17.513 (UTC+0000)] info/network	TCP	| Listening on all interfaces
   [2022-05-30 10:30:17.513 (UTC+0000)] info/network	TCP 4	| Creating server socket for "0.0.0.0" on port 4840
   [2022-05-30 10:30:17.513 (UTC+0000)] info/network	TCP 5	| Creating server socket for "::" on port 4840
   [2022-05-30 10:30:17.514 (UTC+0000)] info/network	TCP 6	| Creating server socket for "127.0.1.1" on port 4840
   [2022-05-30 10:30:17.514 (UTC+0000)] warn/network	TCP 6	| Error binding the socket to the address 127.0.1.1 (Address already in use)

closed 2022-06-07 by Fabrizio997

Vermounthlly, 2022-08-19:
> the past problem with ([2022-05-26 16:08:37.636 (UTC+0000)] warn/network TCP | getaddrinfo lookup for "fabrizio" on port 4840 failed (None)) is no more there but now the when I launch the server it [...]

@Fabrizio997 
I also have the same problem, did you solve the problem？