open62541/open62541 #5154: Сlient reads wrong value

strager14, 2022-05-27:
I have a server that hosts a lot of dynamically changing float variables. I read them with the client in cycle
```
status = UA_Client_readValueAttribute(m_client, UA_NODEID_STRING(uaIndex, cNodeId), &value);

if (status == UA_STATUSCODE_GOOD && UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_FLOAT]))
{
    float fVal = *(UA_Float*)value.data;
}
```

everything works correctly, but sometimes one of the variables gets the value -1.99839716e+18. The server never sends such a value. Moreover, this erroneous value is always the same.

 - [x] open62541 Version (release number or git tag):
 - [x] Operating system: win10 x64

strager14, 2022-05-27:
I figured out that this is happening because the pointer is pointing to dead memory 0xdddddddd

jpfr, 2022-05-27:
Hm. That is strange. This is the first time we get a report with that memory location.
Can you run the unmodified binary with DrMemory to get some log output?
https://drmemory.org/

strager14, 2022-05-30:
> Hm. That is strange. This is the first time we get a report with that memory location. Can you run the unmodified binary with DrMemory to get some log output? https://drmemory.org/
```
UNADDRESSABLE ACCESS beyond heap bounds: reading 0x0000021d4fca4ef2-0x0000021d4fca4ef4 2 byte(s) within 0x0000021d4fca4ef0-0x0000021d4fca4ef4
# 0 opcua.plugin.dll!VariantHasDeadMemoryPtr                          [d:\project\src\plugins\opcua\transportopc.cpp:44]
# 1 opcua.plugin.dll!TransportOPC::SlaveWorkerThread                  [d:\project\src\plugins\opcua\transportopc.cpp:303]
# 2 opcua.plugin.dll!boost::_mfi::mf0<>::operator()                   [d:\project\src\3rdparty\boost\bind\mem_fn_template.hpp:49]
# 3 opcua.plugin.dll!boost::_bi::list1<>::operator()<>                [d:\project\src\3rdparty\boost\bind\bind.hpp:238]
# 4 opcua.plugin.dll!boost::_bi::bind_t<>::operator()                 [d:\project\src\3rdparty\boost\bind\bind.hpp:1273]
# 5 opcua.plugin.dll!boost::detail::thread_data<>::run                [d:\project\src\3rdparty\boost\thread\detail\thread.hpp:120]
# 6 opcua.plugin.dll!boost::`anonymous namespace'::thread_start_function [d:\programming\__c_lib\boost_1_77_0\libs\thread\src\win32\thread.cpp:296]
# 7 ucrtbased.dll!register_onexit_function                           +0xeb     (0x00007ffb70ae4bfc <ucrtbased.dll+0x74bfc>)
# 8 KERNEL32.dll!BaseThreadInitThunk
Note: @0:02:08.460 in thread 28792
Note: next higher malloc: 0x0000021d4fca4f20-0x0000021d4fca4f38
Note: refers to 0 byte(s) beyond last valid byte in prior malloc
Note: prev lower malloc:  0x0000021d4fca4ef0-0x0000021d4fca4ef2
Note: instruction: cmp    (%rax) $0xdddddddd
```

jpfr, 2022-05-30:
Since this is mentioning threading, do you have multithreading activated in the build settings of open62541?
This adds a mutex to protect shared memory.

strager14, 2022-05-30:
> Since this is mentioning threading, do you have multithreading activated in the build settings of open62541? This adds a mutex to protect shared memory.
if you activate multithreading, then the library open62541 does not work at all in my project

strager14, 2022-05-30:
> Since this is mentioning threading, do you have multithreading activated in the build settings of open62541? This adds a mutex to protect shared memory.

I rebuilt from -UA_MULTITTHREADING=101 and in my opinion after that the error began to occur even more often

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier