open62541/open62541 #5156: Session name

OgreTransporter, 2022-05-27:
How to set the session name (https://reference.opcfoundation.org/v105/Core/docs/Part4/5.6.2/#Table15) for the client? I am building an IoT network with OPC. For debugging it would be helpful if I can assign the names of the corresponding devices to the sessions. An output like
```
[2022-05-27 15:51:02.718 (UTC+0200)] info/channel       Connection 496 | SecureChannel 16 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-05-27 15:51:02.718 (UTC+0200)] info/session       SecureChannel 16 | Session "" | Session created
[2022-05-27 15:51:02.718 (UTC+0200)] info/session       SecureChannel 16 | Session "" | ActivateSession: Session activated
```
does not debug well. The server can display/output the name, but how can I set it in the client?

RonnieFTI, 2024-01-25:
I can't get a session to create once connected to ignition.

RudolfRendier, 2025-08-04:
Looks like this PR for 1.4 implements this:
https://github.com/open62541/open62541/pull/6054/commits/25e51495a1df66e5daec086f719c69efc0635ab2

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier