open62541/open62541 #5164: Aes128Sha256RsaOaep does not work with UaExpert

ivan-ushakov, 2022-06-01:
## Description
When I try to connect to server with `Aes128Sha256RsaOaep` and Username/Password I get `Bad_EncodingLimitsExceeded` error.

## Background Information / Reproduction Steps
1. Configure server with `UA_ServerConfig_setDefaultWithSecurityPolicies` function.
2. Try to connect to server with UaExpert, `Aes128Sha256RsaOaep` and Username/Password.

Used CMake options:
```bash
cmake -DCMAKE_BUILD_TYPE=${BUILD_TYPE} -DBUILD_SHARED_LIBS=1 -DUA_LOGLEVEL=200 -DUA_MULTITHREADING=199 -DUA_ENABLE_IMMUTABLE_NODES=1 \
    -DUA_ENABLE_SUBSCRIPTIONS=1 -DUA_ENABLE_METHODCALLS=1 -DUA_ENABLE_PARSING=1 \
    -DUA_ENABLE_NODEMANAGEMENT=1 -DUA_ENABLE_DISCOVERY=1 -DUA_ENABLE_DISCOVERY_MULTICAST=1 \
    -DUA_ENABLE_DISCOVERY_SEMAPHORE=1 -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_TYPEDESCRIPTION=1 -DUA_ENABLE_STATUSCODE_DESCRIPTIONS=1 \
    -DUA_ENABLE_MALLOC_SINGLETON=1 \
    -DUA_ENABLE_PUBSUB=1 -DUA_ENABLE_PUBSUB_ENCRYPTION=1 \
    -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=1 -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=1 \
    -DUA_ENABLE_PUBSUB_DELTAFRAMES=1 -DUA_ENABLE_PUBSUB_FILE_CONFIG=1 -DUA_ENABLE_PUBSUB_MONITORING=1 -DUA_ENABLE_JSON_ENCODING=1 \
    -DUA_ENABLE_PUBSUB_MQTT=1 -DUA_ENABLE_MQTT_TLS=1 -DUA_ENABLE_ENCRYPTION=OPENSSL \
    -DUA_BUILD_EXAMPLES=0 \
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): tags/v1.3
 - [x] Operating system: Debian
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached

```
---
[2022-06-01 15:38:38.910 (UTC+0000)] info/network	Connection 6 | New connection over TCP from ?
[2022-06-01 15:38:39.001 (UTC+0000)] debug/channel	Connection 6 | SecureChannel 4 | Generating new local nonce
[2022-06-01 15:38:39.001 (UTC+0000)] info/channel	Connection 6 | SecureChannel 4 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2022-06-01 15:38:39.061 (UTC+0000)] debug/session	SecureChannel 4 | Session "" | Processing GetEndpointsRequest with endpointUrl "opc.tcp://?:?/"
[2022-06-01 15:38:39.061 (UTC+0000)] debug/session	SecureChannel 4 | Session "" | Sending response for RequestId 2 of type GetEndpointsResponse
[2022-06-01 15:38:39.190 (UTC+0000)] info/channel	Connection 6 | SecureChannel 4 | CloseSecureChannel
[2022-06-01 15:38:39.190 (UTC+0000)] info/network	Connection 6 | Closed
```

mlgiraud, 2022-06-02:
Potential fix in #5168
Please verify and close if this fixes your issue. Otherwise leave a comment and i will further investigate.

ivan-ushakov, 2022-06-02:
> Potential fix in #5168
> Please verify and close if this fixes your issue. Otherwise leave a comment and i will further investigate.

I use OpenSSL implementation

ivan-ushakov, 2022-06-07:
@mlgiraud
Sorry for pushing, but maybe you could tell me the direction I can research this problem myself?

NoelGraf, 2022-12-12:
@ivan-ushakov sorry for the late reply.
I couldn't reproduce the issue. Does the issue still exist?
Otherwise, please provide additional information.

NoelGraf, 2023-07-25:
Close the issue for now, reopen it if necessary.

closed 2023-07-25 by NoelGraf