open62541/open62541 #5170: Incorrect cast UA_String

strager14, 2022-06-03:
i am trying to cast the value read by the client to a UA_STRING
```
UA_Client_readValueAttribute(m_client, UA_NODEID_STRING(uaIndex, cNodeId), &value);
if (UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_STRING]))
{
    UA_String uaStr = *(UA_String*)value.data;
}
```
however, in the end uaStr contains the desired string followed by garbage. In this case, the length of the string is determined correctly. For example, the uaStr.length is 10, but the uaStr.data contains 20 characters. If you try to cut them off, then sooner or later it will lead to a fall.
 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Operating system: win64

mlgiraud, 2022-06-03:
Srings in opcua are not null-terminated. You will always have to use the length supplied in uaStr.length.

jpfr, 2022-06-09:
this kind of issue appears too often imho.
Whether this isn’t our fault or not.

What we could do going forward is to null terminate our strings in addition to providing an explicit length.
That could make the interfacing with old style C libraries easier.
And we can eliminate a whole class of potential bugs.

on the other hand we need to allocate a single byte for empty (but not null) strings.

what do you think @mlgiraud ?