open62541/open62541 #5183: CI build of pull requests: How to re-run failed checks?

keba-estr, 2022-06-09:
## Description

Is there a possibility to restart a single check in the CI build of a pull request, if it failed because of a sporadic error (for example: download of some debian archive failed because of a connection timeout)?

The only way to restart the checks I found so far is to push again a changed commit.

GoetzGoerisch, 2022-06-09:
As far I am aware only the maintainers could rerun failed test manually for this repo.

For your fork, you can enable the GitHub Actions to also run in the fork
[image]

There your have the controls as well to re run failed runs.

keba-estr, 2022-06-09:
Thank you very much for the detailed answer.

closed 2022-06-09 by keba-estr