open62541/open62541 #5198: Client disconnect with BadSecurityChecksFailed when deleting subscriptions (MacOS on M1 only)

basyskom-jvoe, 2022-06-13:
## Description
I am currently attempting to update Qt OPC UA to open62541 1.3.1 to get access to the new client side session localeIds API.
The Qt OPC UA tests use an open62541 based test server and a client which makes calls to this server.

During our integration attempts, we have encountered a very strange issue which only appears on MacOS running on M1 hardware:
When subscriptions are deleted from the server, it sends PublishResponse messages with status code BadNoSubscription for all outstanding publish requests. Sometimes during this process, the client closes the connection with status code BadSecurityChecksFailed:

```qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received a ServiceFault response"
qt.opcua.plugins.open62541.sdk.client: "The ServiceResult has the StatusCode BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Received Publish Response with code BadNoSubscription"
qt.opcua.plugins.open62541.sdk.client: "Error receiving the response with status code BadSecurityChecksFailed"
qt.opcua.plugins.open62541.sdk.securechannel: "Connection 24 | SecureChannel 3 | Receiving the response failed with StatusCode BadSecurityChecksFailed"
```
I have traced the disconnect to https://github.com/open62541/open62541/blob/v1.3.1/src/client/ua_client.c#L252
I see two possible causes:
- The server sends a PublishResponse it should not send
- Somehow, the client side callback is removed before the PublishResponse arrives

All other platforms show the BadNoSubscription messages (as expected) but only MacOS on M1 hardware gets the disconnect.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.3.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: MacOS 12.4
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-01-30:
@basyskom-jvoe Does this happen on master as well?
Please run with valgrind to see if it is a memroy issue...

basyskom-jvoe, 2024-09-19:
The issue seems to be gone with 1.4

closed 2024-09-19 by basyskom-jvoe