open62541/open62541 #5214: Issue 4338 revoked with version v1.3.1?

ffoehlisch, 2022-06-22:
I am trying top compile open62541 on ubuntu with:

cmake -DCMAKE_INSTALL_PREFIX=/home/ff69024/src/open62541/wsbuild/install -DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_WEBSOCKET_SERVER=ON -DLIBWEBSOCKETS_INCLUDE_DIR=.../src/libwebsockets/build/install/include -DLIBWEBSOCKETS_LIBRARIES=.../src/libwebsockets/build/install//lib/libwebsockets.a -DUA_ENABLE_JSON_ENCODING=ON ..

N.b. I am using version v4.2.0 of libwebsockets installed in .../src/libwebsockets/build/install.

Is it possble that the [Issue 4338](https://github.com/open62541/open62541/issues/4338) has been revoked?

In network_ws.c this call:
```
    info.ws_ping_pong_interval = 10;
```
gives a compilation error. 

According to [this commit](https://github.com/warmcat/libwebsockets/commit/280c81278dcb88c074aa19cbf8de809e366a9ece) this field is not required in libwesocket for quiet a while. 

Commenting the line out allows to build libopen62541. However I didn't test it to be working so far...

nullr0ute, 2023-01-24:
I'm still seeing this with 1.3.4 on Linux using libwebsockets 4.3.2 too

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier