open62541/open62541 #5226: Initial data change callback is sometimes not called

basyskom-jvoe, 2022-06-27:
## Description
Sometimes when creating a monitored item using the synchronous open62541 client API, the initial data change notification is dropped with the following message:
```
Could not process a notification with clienthandle 9 on subscription 4
```
The error message is printed in https://github.com/open62541/open62541/blob/1.3/src/client/ua_client_subscriptions.c#L917 and indicates that the client's monitored item context has not yet been inserted into the list when the notification arrives.

The issue occurs with open62541 1.2 and 1.3, the server is also based on open62541 (same version).

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

JulZimmermann, 2024-01-10:
I have the same problem when creating monitored items via the async api.
I create monitored items dynamically while the application is running. `UA_Client_run_iterate` runs in a loop in an own thread.

The calls to `UA_Client_Subscriptions_create_async`, `UA_Client_MonitoredItems_createDataChanges_async`, and `UA_Client_run_iterate` are protected by a mutex.

I analyzed the network communication and it seems like the flow is the fallowing:
1. CreateMonitoredItemsRequest is send by the client
2. PublishRequest is send by the client
3. PublishResponse is send by the server, already containing the initial values of the requested monitored items
4. CreateMonitoredItemsResponse is send by the server

I'm not sure if the server is allowed to send the values inside the PublishResponse before sending the CreateMonitoredItemsResponse, or if the the OPCUA standard is violated by that. Anyway, the open62541 client lib can't handle this and will discard the values.

basyskom-jvoe, 2024-11-15:
I have instrumented some places in the code with marker output and according to my analysis, the `CreateMonitoredItemsResponse` arrives before the `PublishResponse` with the initial value (as expected).

```
qt.opcua.plugins.open62541.sdk.client: Decode a message of type CreateMonitoredItemsResponse
qt.opcua.plugins.open62541.sdk.client: Decode a message of type PublishResponse
qt.opcua.plugins.open62541.sdk.client: Got a notification with clienthandle 2 on subscription 1
qt.opcua.plugins.open62541.sdk.client: Could not process a notification with clienthandle 2 on subscription 1
qt.opcua.plugins.open62541.sdk.client: Response for synchronous CreateMonitoredItem processed: 2
```

The root of the problem stated in the initial report seems to be that the call to `el->run()` in `__Client_Service()` sometimes still runs when the `PublishResponse` arrives and calls the publish response processing code.
At that time, `ua_Client_MonitoredItems_create()` is still blocked by the call to `__Client_Service()` and has not yet been able to add the local context information via `ua_MonitoredItems_create()`.

@jpfr Is there some event loop magic we could use to fix the issue? For example something to stop the `run()` after a certain synchronous response has been received?
Handling the local context information creation directly when the `CreateMonitoredItemsResponse` is received doesn't help because we would have everything to invoke the data change callback, but the caller of `UA_Client_MonitoredItems_createDataChanges()` is still waiting for the return and has not yet received the new monitored item id.

jpfr, 2024-11-15:
Hmm okay.
I understand it as follows.

The client sends a CreateMonitoredItemsRequest.
The server answers with a CreateMonitoredItemsResponse and also with a PublishResponse at the same time.

The client processes both responses before returning to the user.

What we need (imho) is for the synchronous message processing to only process responses until the synchronous response we have been waiting for. The remaining bytes should be stored and only processed in the next call to the EventLoop.

This requires some rework in the SecureChannel. But should be doable.

jpfr, 2024-11-16:
The fix is in #6878.
Do you have a way to verify?
More unit tests are still missing.

basyskom-jvoe, 2024-11-18:
@jpfr Many thanks, this looks exactly like what we need to fix the issue!

Unfortunately I have no way to reproduce the issue with every run. But usually 1 in 3 CI runs will go wrong, so I'll just run it 10 times and report back.

basyskom-jvoe, 2024-11-18:
@jpfr Looks good for the first runs, the issue hasn't occurred again for now...

Integrating the next 1.4 release in Qt OPC UA should take care of our problems, but this will require some work on our side to re-add support the  the Basic256 and Basic128Rsa15 policies which have been commented out in `UA_ClientConfig_setDefaultEncryption()` between v1.4.6 and v1.4.7 because there are still old servers in the field which don't support anything newer.

jpfr, 2024-11-18:
These old SecurityPolicies are still there but commented out.
They are deprecated for a reason.
There are working exploits (against any OPC UA server) that break these encryption schemes within minutes.

jpfr, 2024-11-18:
You're welcome 👍

commit 155dfb06 (2024-11-17): fix(core): Stop processing after a synchronous response

Before it was possible that two responses were received at the same
time. If the client is waiting for the first response, then the second
was processed also before returning the control flow to the client.

This lead to situations where a PublishResponse was processed before
notifying the client about a MonitoredItem handle established just
before.

This fixes #5226.