open62541/open62541 #5227: Chained servers feasibility consulting

zxx8808, 2022-06-28:
## Description

zxx8808, 2022-06-30:
> Yes, this is indeed feasible. It is even described in the standard as an “aggregation server”. Don’t underestimate the many edge cases that have to be covered. For example if the information model changes at runtime.

The nodes of the local layer are modeled or created dynamically by code, some object types and their objects, the values of the variables in the objects come from an xml configuration file. 

Then I want to ask how to model the central layer, which is the second step mentioned above, I have learned several methods, please advise which is the most suitable, or do you have any suggestions.

- Method 1: The central layer also creates instances through an aggregated xml configuration file, just like the local layer, because both central and local services are developed by ourselves;
- Method 2: The central layer dynamically obtains the list of nodes in the local layer using the browse service, and then dynamically generates nodes.
- Method 3: The central layer is not modeled, it is just a transparent proxy layer, and there is no node list like the local layer inside. But because I want pubsub to the external MQTT broker, there must be nodes, which is not the recommended way, is it?

Thank you for your reply!

marwinglaser, 2025-02-25:
Is this still relevant or can this be closed?

o6-nbeier, 2026-10-20:
Thank you for the question.

This is a valid aggregation setup: the central server also acts as a client, imports or shares the model, feeds values via subscriptions or callbacks, forwards commands explicitly and publishes the selected variables via MQTT PubSub.

Closing this for now.

closed 2026-10-20 by o6-nbeier