open62541/open62541 #5236: FreeBSD 13.1: Implicit declaration of pipe2

herrhotzenplotz, 2022-07-05:
## Description

```console
$ git show | head -1
commit 8c64fabafd6f279bff77e7a9a4261d82189e816c
$ uname -a
FreeBSD hades.herrhotzenplotz.de 13.1-RELEASE FreeBSD 13.1-RELEASE releng/13.1-n250148-fc952ac2212 GENERIC amd64
$ freebsd-version -kru
13.1-RELEASE
13.1-RELEASE
13.1-RELEASE
$ 
```

Fails with:
```console
open62541/arch/eventloop_posix_interrupt.c:364:15: error: implicit declaration of function 'pipe2' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    int err = pipe2(pipefd, O_NONBLOCK);
              ^
open62541/arch/eventloop_posix_interrupt.c:364:15: note: did you mean 'pipe'?
/usr/include/unistd.h:358:6: note: 'pipe' declared here
int      pipe(int *);
         ^
```

Maybe not as C99-ish as claimed :-)

## Background Information / Reproduction Steps

Used CMake options:

```console
$ cmake CC=/usr/bin/cc CXX=/usr/bin/c++ -S. -Bbuild-freebsd131-amd64 -GUnix\ Makefiles -DCMAKE_INSTALL_PREFIX=/opt/sn
$ cpuset -l2-3 make -Cbuild-freebsd131-amd64 -j 2 VERBOSE=1 2>&1 | tee ../open62541.build-boom.log
```

## Notes

Sort of a hack, but it fixes the build issues:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index af36586b..07c97854 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -688,6 +688,9 @@ endfunction()
 
 if(NOT UA_FORCE_CPP AND (CMAKE_COMPILER_IS_GNUCC OR CMAKE_C_COMPILER_ID STREQUAL "Clang"))
     check_add_cc_flag("-std=c99")   # C99 mode
+    if(${CMAKE_SYSTEM_NAME} MATCHES "FreeBSD")
+        check_add_cc_flag("-D__BSD_VISIBLE=1")
+    endif()
     check_add_cc_flag("-pipe")      # Avoid writing temporary files (for compiler speed)
     check_add_cc_flag("-Wall")      # Warnings
     check_add_cc_flag("-Wextra")    # More warnings
```

There might be a better solution which I am not aware of.

jpfr, 2022-07-05:
The proposed patch looks good.
You want to send us a pull request?

herrhotzenplotz, 2022-07-06:
I just noticed that defining `_XOPEN_SOURCE` to 600 fixes the issue too. This is imo a cleaner solution. I will adapt the PR.

autoantwort, 2022-10-19:
FYI I get the same error on macos

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

pipe2 was replaced by pipe plus explicit non-blocking setup in #5238, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier