open62541/open62541 #5254: Check input/output arguments of a servermethod from Client

oriolorra, 2022-07-12:
## Description
Hi everyone,

Is there any way to check which arguments have a method from an OPC UA Client?

Thanks in advance

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag): v1.3.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

oriolorra, 2022-07-13:
I do not know how to do that. I am pretty new in the OPC UA world. 
I have done this on client side to browse variables/methods.
<code>

    UA_BrowseRequest bReq;

    UA_BrowseRequest_init(&bReq);

    bReq.requestedMaxReferencesPerNode = 0;

    bReq.nodesToBrowse = UA_BrowseDescription_new();

    bReq.nodesToBrowseSize = 1;

    bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER); /* browse objects folder */

    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; /* return everything */

    UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

    printf("\n%-9s | %-16s | %-16s | %-16s | %-24s | %-24s\n", "NAMESPACE", "NODEID", "BROWSE NAME",
           "DISPLAY NAME", "NODE CLASS", "DATA TYPE");

    for(size_t i = 0; i < bResp.resultsSize; ++i) {

        for(size_t j = 0; j < bResp.results[i].referencesSize; ++j) {

            UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
            std::string node_clas_str = GetNodeClassAsString(ref->nodeClass);

            if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {

                /* NodeId of the variable holding the current time */
                UA_Variant value; /* Variants can hold scalar values and arrays of any type */
                UA_Variant_init(&value);
                const UA_NodeId nodeId = UA_NODEID_NUMERIC(ref->nodeId.nodeId.namespaceIndex, ref->nodeId.nodeId.identifier.numeric);
                UA_Client_readValueAttribute(client, nodeId, &value);
                if(value.type != NULL) 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                           ref->nodeId.nodeId.namespaceIndex,
                           ref->nodeId.nodeId.identifier.numeric,
                           (int)ref->browseName.name.length, ref->browseName.name.data,
                           (int)ref->displayName.text.length, ref->displayName.text.data,
                           (int)strlen(node_clas_str.c_str()), (char*)node_clas_str.c_str(),
                           (int)strlen(value.type->typeName), (char*)value.type->typeName);
                }
                else 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                           ref->nodeId.nodeId.namespaceIndex,
                           ref->nodeId.nodeId.identifier.numeric,
                           (int)ref->browseName.name.length, ref->browseName.name.data,
                           (int)ref->displayName.text.length, ref->displayName.text.data,
                           (int)strlen(node_clas_str.c_str()), (char *)node_clas_str.c_str(),
                           0, "");
                }
            } else if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING) {

                /* NodeId of the variable holding the current time */
                UA_Variant value; /* Variants can hold scalar values and arrays of any type */
                UA_Variant_init(&value);
                const UA_NodeId nodeId = UA_NODEID_STRING(ref->nodeId.nodeId.namespaceIndex, (char *)ref->nodeId.nodeId.identifier.string.data);
                UA_Client_readValueAttribute(client, nodeId, &value);
                if(value.type != NULL) 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                            ref->nodeId.nodeId.namespaceIndex,
                            ref->nodeId.nodeId.identifier.numeric,
                            (int)ref->browseName.name.length, ref->browseName.name.data,
                            (int)ref->displayName.text.length, ref->displayName.text.data,
                            (int)strlen(node_clas_str.c_str()), (char *)node_clas_str.c_str(),
                            (int)strlen(value.type->typeName), (char *)value.type->typeName);
                } 
                else 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                           ref->nodeId.nodeId.namespaceIndex,
                           ref->nodeId.nodeId.identifier.numeric,
                           (int)ref->browseName.name.length, ref->browseName.name.data,
                           (int)ref->displayName.text.length, ref->displayName.text.data,
                           (int)strlen(node_clas_str.c_str()), (char *)node_clas_str.c_str(),
                           0, "");
                }
            }
            /* TODO: distinguish further types */
        }
    }
    UA_BrowseRequest_clear(&bReq);
    UA_BrowseResponse_clear(&bResp);</code>

Getting this output:
NAMESPACE | NODEID           | BROWSE NAME      | DISPLAY NAME     | NODE CLASS               | DATA TYPE               
0         | 61               | FolderType       | FolderType       | UA_NODECLASS_OBJECTTYPE  |                         0         | 23470            | Aliases          | Aliases          | UA_NODECLASS_OBJECT      |                         0         | 2253             | Server           | Server           | UA_NODECLASS_OBJECT      |                         1         | 10               | the answer       | the answer       | UA_NODECLASS_VARIABLE    | Int32                   1         | 62541            | hello world      | Hello World      | UA_NODECLASS_METHOD      |                         1         | 62542            | Turn LED on/of   | Turn LED on/off  | UA_NODECLASS_METHOD      |

oriolorra, 2022-07-13:
I do not know how to do that. I am pretty new in the OPC UA world. 
I have done this on client side to browse variables/methods.
<code>

    UA_BrowseRequest bReq;

    UA_BrowseRequest_init(&bReq);

    bReq.requestedMaxReferencesPerNode = 0;

    bReq.nodesToBrowse = UA_BrowseDescription_new();

    bReq.nodesToBrowseSize = 1;

    bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER); /* browse objects folder */

    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; /* return everything */

    UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

    printf("\n%-9s | %-16s | %-16s | %-16s | %-24s | %-24s\n", "NAMESPACE", "NODEID", "BROWSE NAME",
           "DISPLAY NAME", "NODE CLASS", "DATA TYPE");

    for(size_t i = 0; i < bResp.resultsSize; ++i) {

        for(size_t j = 0; j < bResp.results[i].referencesSize; ++j) {

            UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
            std::string node_clas_str = GetNodeClassAsString(ref->nodeClass);

            if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {

                /* NodeId of the variable holding the current time */
                UA_Variant value; /* Variants can hold scalar values and arrays of any type */
                UA_Variant_init(&value);
                const UA_NodeId nodeId = UA_NODEID_NUMERIC(ref->nodeId.nodeId.namespaceIndex, ref->nodeId.nodeId.identifier.numeric);
                UA_Client_readValueAttribute(client, nodeId, &value);
                if(value.type != NULL) 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                           ref->nodeId.nodeId.namespaceIndex,
                           ref->nodeId.nodeId.identifier.numeric,
                           (int)ref->browseName.name.length, ref->browseName.name.data,
                           (int)ref->displayName.text.length, ref->displayName.text.data,
                           (int)strlen(node_clas_str.c_str()), (char*)node_clas_str.c_str(),
                           (int)strlen(value.type->typeName), (char*)value.type->typeName);
                }
                else 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                           ref->nodeId.nodeId.namespaceIndex,
                           ref->nodeId.nodeId.identifier.numeric,
                           (int)ref->browseName.name.length, ref->browseName.name.data,
                           (int)ref->displayName.text.length, ref->displayName.text.data,
                           (int)strlen(node_clas_str.c_str()), (char *)node_clas_str.c_str(),
                           0, "");
                }
            } else if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING) {

                /* NodeId of the variable holding the current time */
                UA_Variant value; /* Variants can hold scalar values and arrays of any type */
                UA_Variant_init(&value);
                const UA_NodeId nodeId = UA_NODEID_STRING(ref->nodeId.nodeId.namespaceIndex, (char *)ref->nodeId.nodeId.identifier.string.data);
                UA_Client_readValueAttribute(client, nodeId, &value);
                if(value.type != NULL) 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                            ref->nodeId.nodeId.namespaceIndex,
                            ref->nodeId.nodeId.identifier.numeric,
                            (int)ref->browseName.name.length, ref->browseName.name.data,
                            (int)ref->displayName.text.length, ref->displayName.text.data,
                            (int)strlen(node_clas_str.c_str()), (char *)node_clas_str.c_str(),
                            (int)strlen(value.type->typeName), (char *)value.type->typeName);
                } 
                else 
                {
                    printf("%-9u | %-16u | %-16.*s | %-16.*s | %-24.*s | %-24.*s",
                           ref->nodeId.nodeId.namespaceIndex,
                           ref->nodeId.nodeId.identifier.numeric,
                           (int)ref->browseName.name.length, ref->browseName.name.data,
                           (int)ref->displayName.text.length, ref->displayName.text.data,
                           (int)strlen(node_clas_str.c_str()), (char *)node_clas_str.c_str(),
                           0, "");
                }
            }
            /* TODO: distinguish further types */
        }
    }
    UA_BrowseRequest_clear(&bReq);
    UA_BrowseResponse_clear(&bResp);</code>

Getting this output:

| NAMESPACE  | NODEID  | BROWSE NAME | DISPLAY NAME | NODE CLASS | DATA TYPE |
|---|---|---|---|---|---| 
| 0 | 61 | FolderType | FolderType | UA_NODECLASS_OBJECTTYPE   |   |
| 0 | 23470 | Aliases | Aliases | UA_NODECLASS_OBJECT |   |
| 0 | 2253 | Server | Server | UA_NODECLASS_OBJECT |   |
| 1 | 62541 | the answer | the answer | UA_NODECLASS_VARIABLE | Int32 |
| 1 | 10 | hello world | Hello world | UA_NODECLASS_METHOD |  |

And somehow, is there a way to check which input/output variables types has the 'Hello world' method? example code?

Thanks

o6-nbeier, 2026-10-20:
Thank you for the question.

Browse from the Method node to its `InputArguments` and `OutputArguments` properties and read their Values, instead of reading the Value of the Method itself.

Closing for now.

closed 2026-10-20 by o6-nbeier