open62541/open62541 #5262: NodeSet compiler not working when trying to add custom DataTypes

nnminh171298, 2022-07-15:
## Description
I'm encountering problems when trying to generate code for a custom nodeset + datatypes.  
Snippet I used in CMake:
```
ua_generate_nodeset_and_datatypes(
    NAME "iec61850-7-3"
    FILE_CSV "${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-7-3.NodeSet2.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-7-3.NodeSet2.bsd"
    NAMESPACE_MAP "2:http://opcfoundation.org/UA/IEC61850-7-3"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/IEC61850/Opc.Ua.IEC61850-7-3.NodeSet2.xml"
)

ua_generate_nodeset_and_datatypes(
    NAME "iec61850-6"
    FILE_CSV "${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-6.NodeSet2.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-6.NodeSet2.bsd"
    NAMESPACE_MAP "3:http://opcfoundation.org/UA/IEC61850-6"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/IEC61850/Opc.Ua.IEC61850-6.NodeSet2.xml"
	IMPORT_BSD "UA_TYPES_IEC61850_7_3#${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-7-3.NodeSet2.bsd"
    DEPENDS "iec61850-7-3"
)

ua_generate_nodeset_and_datatypes(
    NAME "iec61850-7-4"
    FILE_CSV "${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-7-4.NodeSet2.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-7-4.NodeSet2.bsd"
    NAMESPACE_MAP "4:http://opcfoundation.org/UA/IEC61850-7-4"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/IEC61850/Opc.Ua.IEC61850-7-4.NodeSet2.xml"
	IMPORT_BSD "UA_TYPES_IEC61850_7_3#${PROJECT_SOURCE_DIR}/../Opc.Ua.IEC61850-7-3.NodeSet2.bsd"
    DEPENDS "iec61850-7-3"
)

add_dependencies(open62541-object open62541-generator-ns-iec61850-7-3)
add_dependencies(open62541-object open62541-generator-ns-iec61850-6)
add_dependencies(open62541-object open62541-generator-ns-iec61850-7-4)
```

Content on my bsd-files:
[Opc.Ua.IEC61850-6.NodeSet2.bsd](https://pastebin.com/pKr6SNrt)
[Opc.Ua.IEC61850-7-3.NodeSet2.bsd](https://pastebin.com/mRZwW6pP)

When trying to compile it, I get the following error message:
```
RuntimeError: Infinite loop detected or type not found while processing types ValueWithUnit: unknonwn subtype ['tnsbase:Unit']. If the unknown subtype is 'Bit', then maybe a struct with optional fields is defined wrong in the .bsd-file. If not, maybe you need to import additional types with the --import flag. E.g. '--import=UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'
```

## Background Information / Reproduction Steps

Used CMake options:

UA_MULTITHREADING=100

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10 64-bit, cross-compiling using gcc-linaro-4.9-2014.11-i686-mingw32_arm-linux-gnueabihf
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2022-08-04:
Is it possible that you still provide your xml and csv file so that I can try to reproduce the error?

nnminh171298, 2022-08-05:
I updated the description with a [link](https://drive.google.com/drive/folders/1VDYZ0VPfjF1k3pYfwx6UftUE63HF-GOm?usp=sharing) to all the files.

Muddasir-Shakil, 2022-09-20:
hi @nnminh171298 
I hope one of these options might help you. I did not looked at your bsd files. But it seems like that you did not define the dependencies properly for data types. It seems like nodesets have internal dependencies for data types. I would suggest to first try to fix your interdependencies with DEPENDS_TYPES (http://www.open62541.org/doc/master/nodeset_compiler.html). 
And then if you have any external dependency you can import with IMPORT_BSD option. 

examples: 

INTERNAL
    IMPORT_BSD "TYPES#${UA_NODESET_DIR}/Schema/Opc.Ua.Types.bsd"

   DEPENDS_TYPES
        "UA_TYPES" "UA_TYPES_DI

closed 2025-08-21 by nnminh171298