open62541/open62541 #5272: Provide proper information for ExtensionObject decoding - 2

SylviaKnof, 2022-07-22:
## Description

In **"Provide proper information for ExtensionObject decoding" #1420** mathkonnert wrote:
"OPC UA 1.04 Part 3 mentions at 5.8, that the Attribute "DataTypeDefinition" is mandatory for structs and enums. This attribute contains a struct of StructureDefinition (for struct datatypes) or EnumDefinition (for enums).

As I understand, this struct contains the information, how the DataType is encoded.

Here is a demo, providing the Structuredefinition for the "Point" datatype. (contained in examples/server_testnodeset.c). With this information, UAExpert is able to decode the extension object.
https://github.com/matkonnerth/open62541/tree/custom_types

For the custom datatype generation with or without nodesecompiler that would mean

add the DataTypeDefinition under BaseDataTypes/Structure or BaseDataTypes/Enumeration (as it is already being done)
add the DataTypeDefinition: would mean to extend UA_DataTypeAttributes, extend the UA_DataTypeNode"

I want to get UaExpert, another OPC UA client from Softing and our customer's OPC UA client to show my user-defined Structure.
It is not possible to share the header files like in the examples.

## Background Information / Reproduction Steps

What I have already done:
- Wrote a Model.xml, used the UA-ModelCompiler to generate the ByteString and put it into my NodeSet.xml
- Added a DefaultBinary section to the NodeSet.xml like the example open62541/open62541/tree/master/examples/nodeset/testnodeset.xml
- generated my server code with nodeset_compiler

But unfortunately UaExpert still shows the same error ("BadDecodingError") and also the Softing OPC UA client ("Invalid encoding byte (0xFF) for NodeId")

[image]

It's the same behaviour like before adding the ByteString into my NodeSet.xml

With WireShark I can see that a malformed packet is sent. That is also the same as before the changes.

[image]

What do I have to do to get this run?
What do you mean exactly with "add the DataTypeDefinition: would mean to extend UA_DataTypeAttributes, extend the UA_DataTypeNode"?
Do I have to change something in my code to get this done?
Is there maybe something else wrong like how I built my user-defined structure in code?

Header:
typedef struct
{
	UA_String MES_PCBID;
	UA_Int32  MES_PCBPosition;
	UA_Int32  MES_PCBStatus;
} MES_PCBData;

static UA_DataTypeMember MES_PCBData_members[3] = {
	/* MES_PCBID */
	{
					UA_TYPENAME("MES_PCBID") /* .memberName */
					UA_TYPES_STRING,         /* .memberTypeIndex, points into UA_TYPES since namespaceZero is true */
					0,                       /* .padding */
					true,                    /* .namespaceZero, see .memberTypeIndex */
					false                   /* .isArray */
	},

	/* MES_PCBPosition */
	{
					UA_TYPENAME("MES_PCBPosition")
					UA_TYPES_INT32, 0, true, false
	},

	/* MES_PCBStatus */
	{
					UA_TYPENAME("MES_PCBStatus")
					UA_TYPES_INT32, 0, true, false
	}
};

Code:
m_PCBDataType = (UA_DataType*)UA_malloc(sizeof(UA_DataType) * 1);
m_PCBDataType[0] =
{
		UA_TYPENAME("PCBDataType")     /* .typeName */
		UA_NODEID_STRING(m_namespaces[3], "Root.Types.DataTypes.BaseDataType.Structure.PCBDataType"),    /* .typeId */
		sizeof(MES_PCBData),                     /* .memSize */
		0,                                                     /* .typeIndex, in the array of custom types */
		UA_DATATYPEKIND_STRUCTURE,   /* .typeKind */
		true,                                                /* .pointerFree */
		false,                                               /* .overlayable (depends on endianness and the absence of padding) */
		3,                                                    /* .membersSize */
		5000,                       /* .binaryEncodingId, the numeric identifier used on the wire (the namespaceindex is from .typeId) */
		MES_PCBData_members
};

m_types = (UA_DataType*)UA_malloc(sizeof(UA_DataType) * 1);
m_types[0] = m_PCBDataType[0];

I can read the node with the structure from my own OPC UA server without any errors.

And I do not understand where the inserted ByteString in the NodeSet.xml is used or where I can see it when the nodeset_compiler has done its job.

And this is very urgent...

Fill out the sections and checklist below (add text at the end of each line).

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0.6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10 (Version 21H2 (Build 19044.1826))
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2022-07-27:
I see that you use v1.06 of open62541.
Is there a chance for you to upgrade to v1.3?

That’s a bit of work initially. 
But the node set compiler was greatly improved in the meantime.

Particularly, I recall that at a time the node set compiler generated raw binary encoding for values internally.
The current node set compiler generates C-structures.

Can you share the generated code privately?
The malformed packet should never happen.
So that would warrant patching old releases…

SylviaKnof, 2022-07-27:
We are trying to upgrade to v1.3.
Could you please send a CMakeLists.txt in order to generate code out of my NodeSet.xml?
What is also necessary to change at my existing server based on version 1.0.6?

SylviaKnof, 2022-07-27:
Is it now possible to generate the namespace and the custom datatypes with CMake using "ua_generate_nodeset_and_datatypes" out of the DataType Definitions in the Nodeset.xml?
Or are there still the csv and the bsd file needed?
If this is the case, version 1.3 is not a possible solution for us because we don't have the csv and bsd files and also no Model.xml.

jpfr, 2022-07-28:
Yes. We can now extract the data type definitions from the xml.
Where the bsd is embedded in a base64 encoding.

@andreasebner and @NoelGraf have the details.

closed 2026-10-20 by o6-nbeier