open62541/open62541 #5276: Inclusion of ua_pubsub_manager.h into ua_server.c

SchutzSeb, 2022-07-27:
## Description

I only use the server without pubsub and to simplify my project I have removed the client and pubsub folders since not needed.
My issue is that there is an include of ua_pubsub_manager.h into ua_server.c without conditional compilation on UA_ENABLE_PUBSUB resulting in compilation error. This include seems not required when pubsub is not used (no warning/error at compilation if I remove it).

Would it be possible to add an ifdef for this include in ua_server.c ?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_ENABLE_PUBSUB=OFF ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-07-28:
Sure. You can send a PR for that.

SchutzSeb, 2022-07-28:
Thanks for your reply jpfr.

I have created the [pull request](https://github.com/open62541/open62541/pull/5278) and will now close the issue.

Best regards.

closed 2022-07-28 by SchutzSeb