open62541/open62541 #5281: cmake build with UA_NAMESPACE_ZERO=FULL fails

augustinust, 2022-07-29:
## Description
Downloaded Version 1.3 as ``tar.gz`` from ``http://www.open62541.org/download/``, tried to run cmake with option
``UA_NAMESPACE_ZERO                FULL``

and got this error message:

```
CMake Error at CMakeLists.txt:1170 (message):
  File
  /.../open62541-1.3/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
  not found.  You probably need to initialize the git submodule for
  deps/ua-nodeset or set open62541_NODESET_DIR.

```

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL ../open62541-1.3/
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

GoetzGoerisch, 2022-07-31:
Due to licensing reasons the UA base _NodeSet_ is not included in the download.

As the error message suggests you need to initialise the git submodule or get the _NodeSet_ from here and specifiy the location as suggested: https://github.com/OPCFoundation/UA-Nodeset/tree/latest/Schema

@andreasebner did I overlooked something?

jpfr, 2022-08-03:
This should be documented better.
And/or we can automatically do the checkout of the submodule via cmake:
https://www.scivision.dev/cmake-git-submodule/

NoelGraf, 2022-08-04:
I have created a PR where the git submodules are automatically initialized via cmake #5290

NoelGraf, 2022-08-08:
@augustinust can the issue be closed?

augustinust, 2022-08-08:
It would be sufficient for me to have it documented as @jpfr suggested.

closed 2023-07-25 by NoelGraf