open62541/open62541 #5283: Compiling for Android

ffreality, 2022-07-31:
Is there any suggestion for android compile ? Documentation said that open62541 supports android but I can not find any android related platform options in cmake.

and is there any limitations about android such as not working features in that platform ? (maybe you can say UA_Encryption doesn't work on android)

jpfr, 2022-08-03:
Android is (mostly) POSIX.
Also I'm pretty sure OpenSSL and MbedTLS both exist for Android.

Make sure that you use the NDK for native binaries.
As we use C and Android apps mostly use Java.

ffreality, 2022-08-03:
> Android is (mostly) POSIX. Also I'm pretty sure OpenSSL and MbedTLS both exist for Android.
> 
> Make sure that you use the NDK for native binaries. As we use C and Android apps mostly use Java.

After cmake step, I was thinking to use VS 2022. Because I am working on Unreal Engine client plugin. Windows version and encryption integrations are ready. Only android one remains.

ffreality, 2022-08-10:
`
Commandline options:
-DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release;MinSizeRel;RelWithDebInfo" -DUA_BUILD_EXAMPLES:BOOL="0" -DUA_ENABLE_JSON_ENCODING:BOOL="1" -DUA_ARCHITECTURE:STRING="posix" -DUA_ENABLE_DISCOVERY:BOOL="1" -DUA_MULTITHREADING:STRING="0" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL="1" -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DUA_ENABLE_HISTORIZING:BOOL="1" -DCMAKE_INSTALL_PREFIX:PATH="C:/Program Files (x86)/open62541" -DUA_ENABLE_DIAGNOSTICS:BOOL="1" -DUA_ENABLE_ENCRYPTION_TPM2:STRING="OFF" -DUA_LOGLEVEL:STRING="300" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL="1" -DUA_ENABLE_AMALGAMATION:BOOL="1" -DUA_ENABLE_PUBSUB:BOOL="1" -DUA_ENABLE_ENCRYPTION:STRING="OFF" 

Cache file:
CMAKE_CONFIGURATION_TYPES:STRING=Debug;Release;MinSizeRel;RelWithDebInfo
UA_BUILD_EXAMPLES:BOOL=0
UA_ENABLE_JSON_ENCODING:BOOL=1
UA_ARCHITECTURE:STRING=posix
UA_ENABLE_DISCOVERY:BOOL=1
UA_MULTITHREADING:STRING=0
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=1
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
UA_ENABLE_HISTORIZING:BOOL=1
CMAKE_INSTALL_PREFIX:PATH=C:/Program Files (x86)/open62541
UA_ENABLE_DIAGNOSTICS:BOOL=1
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=FULL
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=1
UA_ENABLE_AMALGAMATION:BOOL=1
UA_ENABLE_PUBSUB:BOOL=1
UA_ENABLE_ENCRYPTION:STRING=OFF

`

Optional Platform Generator is ARM64 (Also tried ARM)

CMake can create open62541.sln but Visual Studio gives these errors.

`
7>The size of D:/C++/open62541/builds/Android/open62541.c is 11338016 Bytes.
7>Building Custom Rule D:/C++/open62541/CMakeLists.txt
9>------ Build started: Project: open62541-code-generation, Configuration: RelWithDebInfo ARM64 ------
9>Building Custom Rule D:/C++/open62541/CMakeLists.txt
10>------ Build started: Project: open62541-object, Configuration: RelWithDebInfo ARM64 ------
10>Building Custom Rule D:/C++/open62541/CMakeLists.txt
10>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(700,5): error MSB6006: "CL.exe" exited with code 2.
10>cl : command line error D8021: invalid numeric argument '/Wshadow'
10>Done building project "open62541-object.vcxproj" -- FAILED.
11>------ Build started: Project: open62541 (open62541\lib\open62541), Configuration: RelWithDebInfo ARM64 ------
11>Building Custom Rule D:/C++/open62541/CMakeLists.txt
11>LINK : fatal error LNK1181: cannot open input file 'D:\C++\open62541\builds\Android\open62541-object.dir\RelWithDebInfo\open62541.obj'
11>Done building project "open62541.vcxproj" -- FAILED.
12>------ Skipped Build: Project: doc_latex, Configuration: RelWithDebInfo ARM64 ------
12>Project not selected to build for this solution configuration 
13>------ Skipped Build: Project: doc (open62541\doc\doc), Configuration: RelWithDebInfo ARM64 ------
13>Project not selected to build for this solution configuration 
14>------ Skipped Build: Project: clang-tidy, Configuration: RelWithDebInfo ARM64 ------
14>Project not selected to build for this solution configuration 
15>------ Build started: Project: ALL_BUILD, Configuration: RelWithDebInfo ARM64 ------
16>------ Skipped Build: Project: doc_pdf, Configuration: RelWithDebInfo ARM64 ------
16>Project not selected to build for this solution configuration 
15>Building Custom Rule D:/C++/open62541/CMakeLists.txt
17>------ Skipped Build: Project: PACKAGE, Configuration: RelWithDebInfo ARM64 ------
17>Project not selected to build for this solution configuration 
18>------ Skipped Build: Project: INSTALL, Configuration: RelWithDebInfo ARM64 ------
18>Project not selected to build for this solution configuration 
========== Build: 8 succeeded, 2 failed, 0 up-to-date, 8 skipped ==========
`

any suggestion for that ?

ffreality, 2022-08-28:
any suggestion ?

ffreality, 2022-12-02:
solved it
https://github.com/open62541/open62541/issues/5462

closed 2022-12-02 by ffreality