open62541/open62541 #5286: 1.3.3: build fails on linking test binary

kloczek, 2022-08-03:
## Description
Build fails on linking bin/tests/check_subscription_event_filter

## Background Information / Reproduction Steps
cmake settiings:
```console
[tkloczko@devel-g2v x86_64-redhat-linux-gnu]$ cmake -L
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/tkloczko/rpmbuild/BUILD/open62541-1.3.3/x86_64-redhat-linux-gnu" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
-- Cache values
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=/usr/bin/clang-format
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
CMAKE_INSTALL_PREFIX:PATH=/usr
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=ON
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_ENCRYPTION:STRING=ON
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED
```

And build fails on:
```console
Consolidate compiler generated dependencies of target check_subscription_event_filter
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/open62541-1.3.3/x86_64-redhat-linux-gnu'
make  -f tests/CMakeFiles/check_subscription_event_filter.dir/build.make tests/CMakeFiles/check_subscription_event_filter.dir/build
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/open62541-1.3.3/x86_64-redhat-linux-gnu'
[ 75%] Linking C executable ../bin/tests/check_subscription_event_filter
cd /home/tkloczko/rpmbuild/BUILD/open62541-1.3.3/x86_64-redhat-linux-gnu/tests && /usr/bin/cmake -E cmake_link_script CMakeFiles/check_subscription_event_filter.dir/link.txt --verbose=1
/usr/bin/gcc -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -std=c99 -fPIC -DNDEBUG -Wl,-z,relro -Wl,--as-needed -Wl,--gc-sections -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin -Wl,--build-id=sha1 CMakeFiles/check_subscription_event_filter.dir/server/check_subscription_event_filter.c.o "../CMakeFiles/open62541-object.dir/src/ua_types.c.o" "../CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o" "../CMakeFiles/open62541-object.dir/src/ua_types_print.c.o" "../CMakeFiles/open62541-object.dir/src_generated/open62541/types_generated.c.o" "../CMakeFiles/open62541-object.dir/src_generated/open62541/transport_generated.c.o" "../CMakeFiles/open62541-object.dir/src_generated/open62541/statuscodes.c.o" "../CMakeFiles/open62541-object.dir/src/ua_util.c.o" "../CMakeFiles/open62541-object.dir/src/ua_timer.c.o" "../CMakeFiles/open62541-object.dir/src/ua_connection.c.o" "../CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o" "../CMakeFiles/open62541-object.dir/src/ua_securechannel_crypto.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_session.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_ns0.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_ns0_diagnostics.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_config.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_utils.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_discovery.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_server_async.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_networkmessage.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_writer.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_writergroup.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_reader.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_readergroup.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_manager.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_ns0.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_method.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_monitoreditem.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_services_discovery_multicast.c.o" "../CMakeFiles/open62541-object.dir/src/client/ua_client.c.o" "../CMakeFiles/open62541-object.dir/src/client/ua_client_connect.c.o" "../CMakeFiles/open62541-object.dir/src/client/ua_client_discovery.c.o" "../CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel.c.o" "../CMakeFiles/open62541-object.dir/src/client/ua_client_subscriptions.c.o" "../CMakeFiles/open62541-object.dir/deps/libc_time.c.o" "../CMakeFiles/open62541-object.dir/deps/pcg_basic.c.o" "../CMakeFiles/open62541-object.dir/deps/base64.c.o" "../CMakeFiles/open62541-object.dir/deps/aa_tree.c.o" "../CMakeFiles/open62541-object.dir/deps/ziptree.c.o" "../CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_config.c.o" "../CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o" "../CMakeFiles/open62541-object.dir/src/ua_types_lex.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_subscription.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_subscription_monitoreditem.c.o" "../CMakeFiles/open62541-object.dir/src/server/ua_subscription_datachange.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/securitypolicy_openssl_common.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_basic128rsa15.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_basic256.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_basic256sha256.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_aes128sha256rsaoaep.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_openssl_create_certificate.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/openssl/ua_pki_openssl.c.o" "CMakeFiles/open62541-testplugins.dir/__/arch/network_tcp.c.o" "CMakeFiles/open62541-testplugins.dir/testing-plugins/testing_clock.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/ua_log_stdout.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/ua_config_default.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/ua_accesscontrol_default.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/ua_nodestore_ziptree.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/ua_nodestore_hashmap.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/ua_securitypolicy_none.c.o" "CMakeFiles/open62541-testplugins.dir/__/plugins/crypto/ua_pki_none.c.o" "CMakeFiles/open62541-testplugins.dir/testing-plugins/testing_policy.c.o" "CMakeFiles/open62541-testplugins.dir/testing-plugins/testing_networklayers.c.o" "CMakeFiles/open62541-testplugins.dir/__/arch/posix/ua_architecture_functions.c.o" -o ../bin/tests/check_subscription_event_filter  -lcheck /usr/lib64/libssl.so /usr/lib64/libcrypto.so -lm -lpthread -lrt -lsubunit
/usr/bin/ld: /tmp/ccGAVOt1.lto.o: in function `eventSetup':
[... cut ...]
make[1]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/open62541-1.3.3/x86_64-redhat-linux-gnu'
make: *** [Makefile:169: all] Error 2
```
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.3
 - [ ] Other OPC UA SDKs used (client or server): N/A
 - [ ] Operating system: Linux x86/64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached N/A
 - [ ] Wireshark network dump attached N/A
 - [ ] Self-contained code example attached N/A
 - [ ] Critical issue

kloczek, 2022-08-26:
gentle ping .. 😋

marwinglaser, 2025-02-18:
I can not reproduce on 1.4 or master. Closing.

closed 2025-02-18 by marwinglaser