open62541/open62541 #5287: pointerfree types in nodeset generator

keba-estr, 2022-08-03:
## Description

The following line 94 in tools/nodeset_compiler/type_parser.py seems to be wrong:
        self.pointerfree = self.name in builtin_pointerfree

Was the following intended?
        self.pointerfree = True in builtin_pointerfree

## Checklist

 - [ ] open62541 Version (release number or git tag): 1.3.2

andreasebner, 2022-08-04:
Please check the issue  @NoelGraf

NoelGraf, 2022-08-04:
Looks correct to me. It checks if the name of the type is in the builtin_pointerfree list and returns true or false accordingly.

keba-estr, 2022-08-04:
Ok, I read it as
(self.pointerfree = self.name) in builtin_pointerfree
but it is actually
self.pointerfree = (self.name in builtin_pointerfree)
So it is ok.

closed 2022-08-04 by keba-estr