open62541/open62541 #5295: Werror=deprecated-declarations since openSSL 3.0

Boberic1999, 2022-08-08:
## Description
I'm trying to write a server with OPENSSL encryption (x509 certificates). I would use UaExpert as client. After enabling openSSL encryption, and building, im getting output in ubuntu 22.04 terminal :

/open62541-1.3.3/build/open62541.c: In function ‘UA_Openssl_RSA_Private_Decrypt’:

/open62541-1.3.3/build/open62541.c:61204:5: error: ‘RSA_size’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]

61204 |     UA_Int32 keySize = RSA_size(get_pkey_rsa(privateKey));

      |     ^~~~~~~~

In file included from /usr/include/openssl/x509.h:36,

                 from /home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:60832:

/usr/include/openssl/rsa.h:204:27: note: declared here

  204 | OSSL_DEPRECATEDIN_3_0 int RSA_size(const RSA *rsa);

      |                           ^~~~~~~~

/home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:61204:5: error: ‘EVP_PKEY_get0_RSA’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]

61204 |     UA_Int32 keySize = RSA_size(get_pkey_rsa(privateKey));

      |     ^~~~~~~~

In file included from /usr/include/openssl/x509.h:29,

                 from /home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:60832:

/usr/include/openssl/evp.h:1346:22: note: declared here

 1346 | const struct rsa_st *EVP_PKEY_get0_RSA(const EVP_PKEY *pkey);

      |                      ^~~~~~~~~~~~~~~~~

/home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:61211:9: error: ‘RSA_private_decrypt’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]

61211 |         decryptedBytes = RSA_private_decrypt (keySize,

      |         ^~~~~~~~~~~~~~

In file included from /usr/include/openssl/x509.h:36,

                 from /home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:60832:

/usr/include/openssl/rsa.h:291:5: note: declared here

  291 | int RSA_private_decrypt(int flen, const unsigned char *from, unsigned char *to,

      |     ^~~~~~~~~~~~~~~~~~~

/home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:61214:28: error: ‘EVP_PKEY_get0_RSA’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]

61214 |                            get_pkey_rsa(privateKey),      /* private key      */

      |                            ^~~~~~~~~~~~

In file included from /usr/include/openssl/x509.h:29,

                 from /home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:60832:

/usr/include/openssl/evp.h:1346:22: note: declared here

 1346 | const struct rsa_st *EVP_PKEY_get0_RSA(const EVP_PKEY *pkey);

      |                      ^~~~~~~~~~~~~~~~~

/home/marko/Desktop/najnoviji .zip/open62541-1.3.3/build/open62541.c:60992:27: error: passing argument 4 of ‘RSA_private_decrypt’ discards ‘const’ qualifier from pointer target type [-Werror=discarded-qualifiers]
[... 225 lines cut ...]
sudo apt-get install check libsubunit-dev # for unit tests
sudo apt-get install python-sphinx graphviz # for documentation generation
sudo apt-get install python-sphinx-rtd-theme # documentation style

cd open62541
mkdir build
cd build
cmake ..
make

# select additional features
ccmake -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_ENCRYPTION=OPENSSL ..
make

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): open62541 1.3.3.
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu 22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

danmoser, 2022-08-09:
Same problem here... open62541 1.3 and Ubuntu 22.04

shaohuazhang1, 2022-09-21:
Have you resolved this problem?This issue occur on my linux both 22.04 and 20.04.

shaohuazhang1, 2022-09-21:
openssl -v  openssl: /lib/x86_64-linux-gnu/libcrypto.so.3: version `OPENSSL_3.1.0' not found (required by openssl)

danmoser, 2022-09-21:
@shaohuazhang1 , did you try `apt install openssl` on ubuntu 20.04? It worked for me. 
Alternatively, `apt install libmbedtls-dev` to use TLS library. That lib works on 22.04

shaohuazhang1, 2022-09-22:
Compiled passed!
Tkank you!

shaohuazhang1, 2022-09-22:
What configuration does the OPC server need after enable mbedtls?
configuration on the UAExpert 
security policy Basic128Rsa15
Message Security Mode Sign&Encrypt

shaohuazhang1, 2022-09-22:
And when I compile with gcc -std=c99 open62541.c tutorial_pubsub_connection.c 
/usr/bin/ld: open62541.c:(.text+0x441052): undefined reference to `mbedtls_rsa_rsaes_oaep_encrypt'
/usr/bin/ld: /tmp/ccdNmZep.o: in function `mbedtls_decrypt_rsaOaep':
open62541.c:(.text+0x44114a): undefined reference to `mbedtls_rsa_set_padding'
/usr/bin/ld: open62541.c:(.text+0x4411f1): undefined reference to `mbedtls_ctr_drbg_random'
/usr/bin/ld: open62541.c:(.text+0x4411f9): undefined reference to `mbedtls_rsa_rsaes_oaep_decrypt'
/usr/bin/ld: /tmp/ccdNmZep.o: in function `UA_mbedTLS_LoadPrivateKey':
open62541.c:(.text+0x4412fe): undefined reference to `mbedtls_pk_parse_key'
/usr/bin/ld: /tmp/ccdNmZep.o: in function `UA_mbedTLS_LoadLocalCertificate':
open62541.c:(.text+0x44137e): undefined reference to `mbedtls_x509_crt_init'
/usr/bin/ld: open62541.c:(.text+0x44139e): undefined reference to `mbedtls_x509_crt_parse'
/usr/bin/ld: open62541.c:(.text+0x441421): undefined reference to `mbedtls_x509_crt_free'
collect2: error: ld returned 1 exit status

danmoser, 2022-09-22:
@shaohuazhang1 , when compiling open62541 be sure that you have these options enabled: 
`cmake .. -DUA_ENABLE_ENCRYPTION=ON -DBUILD_SHARED_LIBS=ON -DMBEDTLS_INCLUDE_DIRS=/usr/include/mbedtls`

jpfr, 2022-10-05:
@shaohuazhang1 Please verify and close with the current master.

danmoser, 2022-10-05:
Folks, current master branch compiled here after `apt install libssl-dev openssl` on Ubuntu 22.04. 
Build option: `-DUA_ENABLE_ENCRYPTION=OPENSSL`
Thanks!!

jpfr, 2022-10-05:
Nice! Closing.

closed 2022-10-05 by jpfr

java-code-former, 2022-10-26:
so What is the solution？

jpfr, 2022-10-26:
Use the master branch.
It has support for OpenSSL3.