open62541/open62541 #5297: PubSub MQTT: Connection creation failed

konstihengge, 2022-08-09:
## Description
I'm new here and trying to set up publisher/subscriber communication using MQTT (```tutorial_pubsub_mqtt_publish.c```)
However, there are problems when establishing the connection with the transport layer.
Am I doing something wrong?
I hope you can help me with this...

[image]

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=ON
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2022-08-15:
In the example, the broker address is set to localhost. Do you have an MQTT broker running locally? Otherwise, you can choose a publicly accessible server like test.mosquitto.org.

NoelGraf, 2022-09-08:
@konstihengge Can the issue be closed?

closed 2022-09-13 by NoelGraf