open62541/open62541 #5304: Heap corruption for large data with SignAndEncrypt

jroeglin, 2022-08-17:
## Description
Heap corruption for large data like an image as UA_ByteString with SignAndEncrypt (OpenSSL plugin). I thing the issue also related to [#4515](https://github.com/open62541/open62541/issues/4515) and [#4558](https://github.com/open62541/open62541/issues/4558)

## Background Information / Reproduction Steps
The issue occurs only with SignAndEncrypt. I can transfer my Image with None or Sign. Image size is 24Mb.
I try to configure the server buffer size but its doesn't help. I can transfer small data with SignAndEncrypt like UINT64. Also multiple notes.

Following my Server config:
```cpp
void ConcreteOpcuaServer::setUpServerConfig(int port)
{
    this->setUpBuildInfos();

    this->serverConfig().networkLayersSize = 0;
    
    auto retval = UA_ServerConfig_addNetworkLayerTCP(&this->serverConfig(), port, 0, 0);

    auto cert = ConcreteOpcuaServer::loadFile("C:/certs/server_cert.der");
    auto key = ConcreteOpcuaServer::loadFile("C:/certs/server_key.der");

    size_t trustListSize = 0;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize + 1);
    /* Loading of an issuer list, not used in this application */
    size_t issuerListSize = 0;
    UA_ByteString* issuerList = NULL;
    /* Loading of a revocation list currently unsupported */
    UA_ByteString* revocationList = NULL;
    size_t revocationListSize = 0;
    //
    retval |= UA_CertificateVerification_Trustlist(&this->serverConfig().certificateVerification,
        trustList, trustListSize,
        issuerList, issuerListSize,
        revocationList, revocationListSize);

    //set up Security and Encryption
    retval |= UA_ServerConfig_addSecurityPolicyBasic128Rsa15(&this->serverConfig(), &cert, &key);
    retval |= UA_ServerConfig_addSecurityPolicyBasic256(&this->serverConfig(), &cert, &key);
    retval |= UA_ServerConfig_addSecurityPolicyBasic256Sha256(&this->serverConfig(), &cert, &key);
    retval |= UA_ServerConfig_addSecurityPolicyAes128Sha256RsaOaep(&this->serverConfig(), &cert, &key);

    //allow only security communication
    this->serverConfig().securityPolicyNoneDiscoveryOnly = true;

    //set up login  //TODO Load data from ConfigModule
    logins().resize(1);
    logins()[0].username = UA_STRING_STATIC("admin");
    logins()[0].password = UA_STRING_STATIC("password");
    enableSimpleLogin(false);

    // Adds endpoints for all configured security policies in each mode.
    retval |= UA_ServerConfig_addAllEndpoints(&this->serverConfig());

    if (retval != UA_STATUSCODE_GOOD) {
        UA_ServerConfig_clean(&this->serverConfig());
        throw std::runtime_error("Error: ");
    }
}
...
bool enableSimpleLogin(bool allowAnonymous = true, const std::string& userTokenPolicyUri = "")
{
        ByteString ut(userTokenPolicyUri);
        // install access control into the config that maps on to the server hence its virtual functions
        if (_config->accessControl.clear) {
            _config->accessControl.clear(&_config->accessControl);
        }
        auto retval = UA_AccessControl_default(_config, allowAnonymous, nullptr,
                                  &_config->securityPolicies[_config->securityPoliciesSize-1].policyUri,
                                 _logins.size(), _logins.data());
        setAccessControl(&_config->accessControl);  // map access control requests to this object
        return true;
}

```

Following my Client config:
```cpp
void ConcreteOpcuaClient::callbackConfigDecorator(UA_ClientConfig* config)
{
	auto appUri = UA_String_fromChars("urn:open62541.client.application");
	auto productUri = UA_String_fromChars("http://open62541.org");
	config->clientDescription.applicationUri = appUri;
	config->clientDescription.productUri = productUri;
	
	auto cert = ConcreteOpcuaClient::loadFile("C:/certs/client/client_cert.der");
	auto key = ConcreteOpcuaClient::loadFile("C:/certs/client/client_key.der");
	
	UA_ByteString* revocationList = NULL;
	size_t revocationListSize = 0;
	size_t trustListSize = 0;
	UA_STACKARRAY(UA_ByteString, trustList, trustListSize + 1);
	
	auto retval = UA_CertificateVerification_Trustlist(&config->certificateVerification,
		trustList, trustListSize,
		NULL, 0,
		revocationList, revocationListSize);
	
	config->securityPolicies[0].clear(&config->securityPolicies[0]);
	UA_free(config->securityPolicies);
	config->securityPolicies = 0;
	config->securityPoliciesSize = 0;
	/* Populate SecurityPolicies */
	UA_SecurityPolicy* sp = (UA_SecurityPolicy*)UA_malloc(sizeof(UA_SecurityPolicy)*5);
	config->securityPolicies = sp;
	
	retval |= UA_SecurityPolicy_None(config->securityPolicies,
		cert, &config->logger);
	
	if (retval == UA_STATUSCODE_GOOD) {
		++config->securityPoliciesSize;
	}
	else {
		UA_LOG_WARNING(&config->logger, UA_LOGCATEGORY_USERLAND,
			"Could not add SecurityPolicy#None with error code %s",
			UA_StatusCode_name(retval));
	}

	
	retval = UA_SecurityPolicy_Aes128Sha256RsaOaep(&config->securityPolicies[config->securityPoliciesSize],
		cert, key, &config->logger);
	
	if (retval == UA_STATUSCODE_GOOD) {
		++config->securityPoliciesSize;
	}
	else {
		UA_LOG_WARNING(&config->logger, UA_LOGCATEGORY_USERLAND,
			"Could not add SecurityPolicy#Aes128Sha256RsaOaep with error code %s",
			UA_StatusCode_name(retval));
	}
	
	if (config->securityPoliciesSize == 0) {
		UA_free(config->securityPolicies);
		config->securityPolicies = NULL;
	}
	
	config->securityMode = UA_MessageSecurityMode::UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
}
```

Following the Callstack while the heap corruption occurs:
[image]

Used CMake options:

```bash
-DUA_BUILD_UNIT_TESTS:BOOL="1" -DUA_ENABLE_ENCRYPTION_TPM2:STRING="OFF" -DUA_ARCHITECTURE:STRING="win32" -DCMAKE_BUILD_TYPE:STRING="Debug" -DLIB_EAY_RELEASE:FILEPATH="C:/Program Files/OpenSSL/lib/libcrypto.lib" -DSSL_EAY_DEBUG:FILEPATH="SSL_EAY_DEBUG-NOTFOUND" -DBUILD_SHARED_LIBS:BOOL="1" -DUA_MULTITHREADING:STRING="0" -DUA_ENABLE_HISTORIZING:BOOL="1" -DLIB_EAY_DEBUG:FILEPATH="LIB_EAY_DEBUG-NOTFOUND" -DUA_LOGLEVEL:STRING="300" -DUA_BUILD_TOOLS:BOOL="1" -DOPENSSL_APPLINK_SOURCE:FILEPATH="C:/Program Files/OpenSSL/bin/openssl.exe" -DUA_ENABLE_ENCRYPTION:STRING="OPENSSL" -DCLANG_FORMAT_EXE:FILEPATH="CLANG_FORMAT_EXE-NOTFOUND" -DCMAKE_INSTALL_PREFIX:PATH="C:/dependencies/open62541withoutdll" -DUA_BUILD_EXAMPLES:BOOL="1" -DSSL_EAY_RELEASE:FILEPATH="C:/Program Files/OpenSSL/lib/libssl.lib" -DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release;" 
```
I built open62541 without a DLL because I had problems building it in combination with the cpp wrapper.
Please late me know if you need more information.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.2
 - [X] Other OPC UA SDKs used (client or server):  c++ wrapper [open62541Cpp](https://github.com/bazza2jw/open62541Cpp)
 - [x] Operating system: win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2022-11-14:
This was probably resolved via https://github.com/open62541/open62541/pull/5438.
We made patch releases for 1.2 and 1.3.
Please check if this fixed this issue as well.

NoelGraf, 2022-12-12:
@jroeglin Were you able to check if the issue is fixed?

closed 2023-07-25 by NoelGraf