open62541/open62541 #5306: Zero timestamp in response header for renew secure channel

Rowesca1, 2022-08-22:
## Description
When receiving a renew secure channel message the response header timestamp field is not set.

## Background Information / Reproduction Steps
In the file ua_services_securechannel.c when a new secure channel is opened via UA_SecureChannelManager_open the timestamp is filled in. It is done at line 286.

`response->responseHeader.timestamp = response->securityToken.createdAt; ` 

This is not done when renew is called using UA_SecureChannelManager_renew.
The same line should be added around line 338.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): All
 - [x] Other OPC UA SDKs used (client or server): Server
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-08-30:
Did you have problems with a particular SDK and this?

closed 2022-08-30 by jpfr

Rowesca1, 2022-08-30:
> Did you have problems with a particular SDK and this?

No major problems.

A 3rd party client would generate warnings because of this.

PR #5309 (2022-08-30): fix(server) zero timestamp in response header for renew secure channe…
Fill in the timestamp from the response header for renew secure channel requests.

Might need test?

fixes #5306