open62541/open62541 #5308: PubSub: Preliminary interface for event loop?

keba-estr, 2022-08-22:
## Description

@JanSurft: Is there already a preliminary interface (or feature branch) for the event loop in the PubSub part of the open62541 available?

## Background Information

A preliminary interface would be helpful for developing an task and timing concept for the usage of the event loop in the calling user code.

JanSurft, 2022-09-02:
@keba-estr yes there is a feature branch that integrates the UDP event manager with the eventloop and UDP PubSub: https://github.com/open62541/open62541/pull/5315. If you need any further information other than reviewing the code/diffs yourself come back to me.

keba-estr, 2022-09-02:
Thank you for the information, I will take a look at the code diffs.

closed 2022-09-02 by keba-estr