open62541/open62541 #5310: lds server register stm32

embbo, 2022-08-26:
## Description
I am trying example open62541/examples/discovery/server_register.c
with embedded device stm32 . It complies fines and opcua serstarts . but I am unable to register 
#define DISCOVERY_SERVER_ENDPOINT "opc.tcp://localhost:4840"
should i do something else for embedded device beause it works fine with desktop apllication . 

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Muddasir-Shakil, 2022-09-19:
Hi,
 Do you have LDS server running on the same device?

embbo, 2022-09-22:
@Muddasir-Shakil  
No Client is on the Stm32 and srver is on the Windows using VS2019
I also treid giving ip adress but same it doesnt work.

closed 2026-10-20 by o6-nbeier