open62541/open62541 #5311: PubSub_RealTime example running

OsamaOAyoub, 2022-08-29:
## Description
Hello,

I am trying to run the RT_Publisher example provided by open62541 using a Raspberry PI running PREEMPT-RT, but I am facing several warnings and the example doesn't run and produce the csv file that I need to inspect for performance.

## Background Information / Reproduction Steps
I modified the ETH_ADDRESS and INTERFACE and tried several options.

I run ip addr to get my ethernet interface name and mac address and I used them in pubsub_interrupt_publish.c:

`#define ETH_PUBLISH_ADDRESS      "opc.eth://**-**-**-**-**-**"
#define ETH_INTERFACE            "eth0"`

Stars represent the MAC address

When I run the example I get the following warnings:

`[2022-08-29 13:47:22.463 (UTC+0200)] warn/pubsub        WriterGroup ns=1;i=3    | Writer ns=1;i=4       | PubSub-RT configuration fail: PDS contains field without external data source`

I tried to trace back what does it mean and how I overcome it, but I can't find where should I set the counter value that is being published as UA_VALUEBACKENDTYPE_EXTERNAL

The other warnings that are network related:
`[2022-08-29 13:47:22.464 (UTC+0200)] info/network       TCP     | Listening on all interfaces
[2022-08-29 13:47:22.465 (UTC+0200)] info/network       TCP 5   | Creating server socket for "0.0.0.0" on port 12345
[2022-08-29 13:47:22.465 (UTC+0200)] info/network       TCP 6   | Creating server socket for "::" on port 12345
[2022-08-29 13:47:22.476 (UTC+0200)] info/network       TCP 7   | Creating server socket for "2a01:9700:1184:7500:dea6:32ff:fe1c:ab40" on port 12345
[2022-08-29 13:47:22.477 (UTC+0200)] warn/network       TCP 7   | Error binding the socket to the address 2a01:9700:1184:7500:dea6:32ff:fe1c:ab40 (Address already in use)
[2022-08-29 13:47:22.478 (UTC+0200)] warn/network       TCP 7   | Error binding the socket to the address 192.168.1.23 (Address already in use)
[2022-08-29 13:47:22.479 (UTC+0200)] info/network       TCP 7   | Creating server socket for "192.168.1.24" on port 12345
[2022-08-29 13:47:22.480 (UTC+0200)] warn/network       TCP 7   | Error binding the socket to the address 192.168.1.24 (Address already in use)`

Please note that I tried to keep the port number as default 4840, but it gave me the same warnings.

192.168.1.23: is the ip address of eth0
192.158.1.24: is the ip address of wlan0

I tried to run the example while SSHing through the wifi and the ethernet is disconnected and while the Raspberry PI is connected through the ethernet and I am connected through SSH using the ethernet address.

I am also running it as a root as needed

Used CMake options:

`BUILD_SHARED_LIBS                OFF
 CLANG_FORMAT_EXE                 CLANG_FORMAT_EXE-NOTFOUND
 CMAKE_BUILD_TYPE                 Release
 CMAKE_INSTALL_PREFIX             /usr/local
 LIB_BPF                          LIB_BPF-NOTFOUND
 MDNSD_LOGLEVEL                   300
 UA_ARCHITECTURE                  posix
 UA_BUILD_EXAMPLES                ON
 UA_BUILD_TOOLS                   OFF
 UA_BUILD_UNIT_TESTS              OFF
 UA_ENABLE_AMALGAMATION           OFF
 UA_ENABLE_DA                     ON
 UA_ENABLE_DIAGNOSTICS            ON
 UA_ENABLE_DISCOVERY              ON
 UA_ENABLE_ENCRYPTION             OPENSSL
 UA_ENABLE_ENCRYPTION_TPM2        OFF
 UA_ENABLE_HISTORIZING            OFF
 UA_ENABLE_JSON_ENCODING          OFF
 UA_ENABLE_METHODCALLS            ON
 UA_ENABLE_PUBSUB                 ON
 UA_ENABLE_PUBSUB_DELTAFRAMES     OFF
 UA_ENABLE_PUBSUB_ETH_UADP        ON
 UA_ENABLE_PUBSUB_INFORMATIONMO   OFF
 UA_ENABLE_PUBSUB_INFORMATIONMO   OFF
 UA_ENABLE_SUBSCRIPTIONS          ON
 UA_ENABLE_SUBSCRIPTIONS_EVENTS   ON
 UA_FORCE_WERROR                  ON
 UA_LOGLEVEL                      300
 UA_MULTITHREADING                100
 UA_NAMESPACE_ZERO                FULL
 UA_NODESET_DIR                   /home/ubuntu/open62541/deps/ua-nodeset
 UA_ENABLE_JSON_ENCODING          OFF
 UA_ENABLE_MALLOC_SINGLETON       ON`

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): bf166b5fa2d6f1119724d1bcacef8a7f08c39778
 - [ ] Other OPC UA SDKs used (client or server):  none
 - [x] Operating system: Linux PREEMPT-RT
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

kulveer1998, 2022-10-04:
I still use PubSub example code on raspberry pi , and faced the same issue , did u resolve it ?

OsamaOAyoub, 2022-10-12:
@kulveer1998 I was able to get some results in the csv log by changing 
`#define CLOCKID                  CLOCK_MONOTONIC_RAW`

to

`#define CLOCKID                  CLOCK_REALTIME`

And you can also change 

`#define PUB_INTERVAL             0.25 /* Publish interval in milliseconds */`

To a value that fits your needs, but I can't see the actual counter node in UaExpert 

And I am also still getting the sockets warnings and PDS. Additionally, I am getting this warning at every cycle

`[2022-10-12 03:31:14.936 (UTC-0700)] warn/eventloop	Timeout during poll`

closed 2022-12-22 by OsamaOAyoub