open62541/open62541 #5316: UA_Variant_copy ignores storageType

keba-estr, 2022-08-31:
## Description

If the function UA_Variant_copy is called for a variant which has set the storageType to UA_VARIANT_DATA_NODELETE, the data array pointer should not be allocated and deep copied, but only used from the old data array pointer.

If you confirm that this is the correct usage, I can create a pull request for the change.

The new function would look like this:

`static UA_StatusCode
Variant_copy(UA_Variant const *src, UA_Variant *dst, const UA_DataType *_) {
    size_t length = src->arrayLength;
    if(UA_Variant_isScalar(src))
        length = 1;
    UA_StatusCode retval;
    if(p->storageType == UA_VARIANT_DATA_NODELETE) {
       dst->data = src->data;
       retval = UA_STATUSCODE_GOOD;
    } else {
       retval = UA_Array_copy(src->data, length,
                              &dst->data, src->type);
    }
    if(retval != UA_STATUSCODE_GOOD)
        return retval;
    dst->arrayLength = src->arrayLength;
    dst->type = src->type;
    if(src->arrayDimensions) {
        retval = UA_Array_copy(src->arrayDimensions, src->arrayDimensionsSize,
            (void**)&dst->arrayDimensions, &UA_TYPES[UA_TYPES_INT32]);
        if(retval != UA_STATUSCODE_GOOD)
            return retval;
        dst->arrayDimensionsSize = src->arrayDimensionsSize;
    }
    return UA_STATUSCODE_GOOD;
}
`

## Checklist

 - [ ] open62541 Version (release number or git tag): 1.3.3 (also in latest master with commit e03be588a2a55fcfa0c1f2a00ef0b305c3d0de96)

keba-estr, 2023-07-13:
Create PR #5875 for this issue.

jpfr, 2023-07-13:
> The way UA_VARIANT_DATA_NODELETE is used now is that it only affects the variable data, not the variable arrayDimensions which must be deep copied in both cases.

No. This is what the documentation says:
```c
/* The data is "borrowed" by the variant and is
 * not deleted when the variant is cleared up.
 * The array dimensions also borrowed. */
 ```

In the current code, a normal copy is a deep copy that also copies the array dimensions.
If you make a shallow copy, the value and the array dimensions are re-used, but both are not freed in UA_Variant_clear if the NoDelete flag is still set.

keba-estr, 2023-07-14:
Ok, now I understand the expected behaviour.
My original problem was the following:
For a data variable the type UA_DataValue (including the UA_Variant) is used. Depending whether the address of the data variable is in the same process or not, the memory for the UA_Variant->data is either directly assigned or allocated. In the copy constructor of the data variable I used UA_DataValue_copy to copy the variable. If the memory was directly assigned, the new copied variable has its own buffer and is not updated automatically if the original memory changed.

As I understand it now, I have to check the storage type of the containing variant in the copy constructor outside the open62541 code and do a shallow copy of the UA_DataValue variable in case of storageType UA_VARIANT_DATA_NODELETE, else call UA_DataValue_copy.

closed 2023-07-14 by keba-estr