open62541/open62541 #5319: UA_Client_connectUsername doesn't work in none security mode

tiantao433, 2022-09-02:
## Description

UA_Client_connectUsername failed in none security mode.
But，it is ok in sign and sign&encrypt mode.
other ua client can connect the ua server with username/password in none security mode.

 - [ ] open62541 Version (r1.3.3):
 - [ ] Other OPC UA SDKs used (client):  prosys opc ua client
 - [ ] Operating system:  windows 11
 
## Background Information / Reproduction Steps

Code:
	m_Client = UA_Client_new();
	m_ClientConfig = UA_Client_getConfig(m_Client);
	UA_ClientConfig_setDefault(m_ClientConfig);
	m_ClientConfig->securityMode = UA_MESSAGESECURITYMODE_NONE;
	m_ClientConfig->securityPolicyUri = UA_STRING_NULL;
 	UA_Client_connectUsername(m_Client, m_DbUrl.c_str(), m_DbUser.c_str(), m_DbPasswd.c_str());

Server:  
[image]

Used CMake options:

cmake.exe .. -DUA_ENABLE_ENCRYPTION:STRING="OPENSSL" -G "Visual Studio 17 2022" -A Win32

Log：

[2022-09-02 19:17:34.191 (UTC+0800)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-09-02 19:17:34.196 (UTC+0800)] debug/network      Sent HEL message
[2022-09-02 19:17:34.196 (UTC+0800)] debug/client       Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
[2022-09-02 19:17:34.203 (UTC+0800)] debug/network      Received ACK message
[2022-09-02 19:17:34.203 (UTC+0800)] debug/client       Client Status: ChannelState: AckReceived, SessionState: Closed, ConnectStatus: Good
[2022-09-02 19:17:34.203 (UTC+0800)] debug/channel      Connection 632 | SecureChannel 0 | Generating new local nonce
[2022-09-02 19:17:34.204 (UTC+0800)] debug/channel      Connection 632 | SecureChannel 0 | Requesting to open a SecureChannel
[2022-09-02 19:17:34.204 (UTC+0800)] debug/channel      Requesting to open a SecureChannel
[2022-09-02 19:17:34.205 (UTC+0800)] debug/channel      OPN message sent
[2022-09-02 19:17:34.205 (UTC+0800)] debug/client       Client Status: ChannelState: OPNSent, SessionState: Closed, ConnectStatus: Good
[2022-09-02 19:17:34.208 (UTC+0800)] info/channel       Connection 632 | SecureChannel 100 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-09-02 19:17:34.209 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2022-09-02 19:17:34.209 (UTC+0800)] debug/channel      Connection 632 | SecureChannel 100 | Sending request with RequestId 2 of type GetEndpointsRequest
[2022-09-02 19:17:34.213 (UTC+0800)] info/client        Rejecting endpoint 6: security mode doesn't match
[2022-09-02 19:17:34.213 (UTC+0800)] info/client        Rejecting endpoint 7: security mode doesn't match
[2022-09-02 19:17:34.214 (UTC+0800)] info/client        Rejecting endpoint 8: security mode doesn't match
[2022-09-02 19:17:34.214 (UTC+0800)] info/client        Rejecting endpoint 9: security mode doesn't match
[2022-09-02 19:17:34.214 (UTC+0800)] info/client        Rejecting endpoint 10: security mode doesn't match
[2022-09-02 19:17:34.214 (UTC+0800)] info/client        Rejecting UserTokenPolicy 0 (anonymous) in endpoint 11: configuration doesn't match
[2022-09-02 19:17:34.215 (UTC+0800)] info/client        Rejecting UserTokenPolicy 1 in endpoint 11: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2022-09-02 19:17:34.215 (UTC+0800)] info/client        Rejecting UserTokenPolicy 2 in endpoint 11: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2022-09-02 19:17:34.215 (UTC+0800)] info/client        Rejecting UserTokenPolicy 3 in endpoint 11: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2022-09-02 19:17:34.216 (UTC+0800)] info/client        Rejecting UserTokenPolicy 4 in endpoint 11: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2022-09-02 19:17:34.216 (UTC+0800)] info/client        Rejecting endpoint 12: security mode doesn't match
[2022-09-02 19:17:34.216 (UTC+0800)] info/client        Rejecting endpoint 13: security mode doesn't match
[2022-09-02 19:17:34.216 (UTC+0800)] info/client        Rejecting endpoint 14: security mode doesn't match
[2022-09-02 19:17:34.216 (UTC+0800)] info/client        Rejecting endpoint 15: security mode doesn't match
[2022-09-02 19:17:34.216 (UTC+0800)] info/client        Rejecting endpoint 16: security mode doesn't match
[2022-09-02 19:17:34.216 (UTC+0800)] error/client       No suitable UserTokenPolicy found for the possible endpoints
[2022-09-02 19:17:34.217 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError

sD3E2b, 2022-10-03:
Your server does not accept password as plaintext. You need encryption. See closed issue #4459 

Have a look on this example
`examples/client_connect.c`

tiantao433, 2022-10-08:
Server can accept password as plaintext.  Prosys opc ua client can work properly as plaintext.

tiantao433, 2022-12-13:
I already did this, doesn't work:

	UA_ClientConfig_setDefault(m_ClientConfig);
	m_ClientConfig->securityMode = UA_MESSAGESECURITYMODE_NONE;
	m_ClientConfig->securityPolicyUri = UA_STRING_NULL;

NoelGraf, 2022-12-13:
The policy to be used for authentication is set on the server side. In the function UA_ServerConfig_setDefaultWithSecurityPolicies the function UA_AccessControl_default is called. There the userTokenPolicyUri is passed. Here you can pass e.g. http://opcfoundation.org/UA/SecurityPolicy#None.

On the client side you should set the securityPolicyUri as follows:
`m_ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#None");`

eziosella, 2023-07-03:
Dear all, I have a similar situation.
I am playing with

> example/tutorial_client_firststep.c

a bit modified as below
```
#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>
#include <open62541/plugin/log_stdout.h>

int main(void) {
    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);

    cc->securityMode = UA_MESSAGESECURITYMODE_NONE;
    UA_ByteString_clear(&cc->securityPolicyUri);
    cc->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#None");

    UA_StatusCode retval = UA_Client_connectUsername(client, "opc.tcp://192.168.0.230:4840", "myLogin", "myPassword");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "The connection failed with status code %s",
                    UA_StatusCode_name(retval));
        UA_Client_delete(client);
        return 0;
    }

    /* Read the value attribute of the node. UA_Client_readValueAttribute is a
     * wrapper for the raw read service available as UA_Client_Service_read. */
    UA_Variant value; /* Variants can hold scalar values and arrays of any type */
    UA_Variant_init(&value);

    /* NodeId of the variable holding the current time */
    const UA_NodeId nodeId =
        UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
    retval = UA_Client_readValueAttribute(client, nodeId, &value);

    if(retval == UA_STATUSCODE_GOOD &&
       UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
        UA_DateTime raw_date = *(UA_DateTime *) value.data;
        UA_DateTimeStruct dts = UA_DateTime_toStruct(raw_date);
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "date is: %u-%u-%u %u:%u:%u.%03u",
                    dts.day, dts.month, dts.year, dts.hour,
                    dts.min, dts.sec, dts.milliSec);
    } else {
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "Reading the value failed with status code %s",
                    UA_StatusCode_name(retval));
    }

    /* Clean up */
    UA_Variant_clear(&value);
    UA_Client_delete(client); /* Disconnects the client internally */
    return 0;
}
```
And I get this error on console
```
[2023-07-03 15:01:31.785 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-07-03 15:01:31.785 (UTC+0200)] info/eventloop	Starting the EventLoop
[2023-07-03 15:01:31.785 (UTC+0200)] info/network	TCP 4	| New connection to "192.168.0.230" on port 4840
[2023-07-03 15:01:31.808 (UTC+0200)] info/channel	TCP 4	| SC 53	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-07-03 15:01:31.808 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-07-03 15:01:31.824 (UTC+0200)] info/client	Use the EndpointURL opc.tcp://FAGORCNC:4840 returned from FindServers
[2023-07-03 15:01:31.824 (UTC+0200)] info/channel	TCP 0	| SC 0	| Closed the SecureChannel
[2023-07-03 15:01:31.825 (UTC+0200)] info/network	TCP 5	| New connection to "192.168.0.230" on port 4840
[2023-07-03 15:01:31.825 (UTC+0200)] info/network	TCP 4	| Socket closed
[... 81 lines cut ...]
14:48:40.155 | AddressSpaceModel  | FagorOpcuaServer               | Unregister for ModelChangeEvents returned Good
14:48:40.451 | Server Node        | FagorOpcuaServer               | Connection status of server 'FagorOpcuaServer' changed to 'Disconnected'.
14:48:40.452 | DA Plugin          | FagorOpcuaServer               | Deleting subscription 0
14:48:40.457 | Server Node        | FagorOpcuaServer               | Disconnect succeeded.
14:49:45.037 | Server Node        | FagorOpcuaServer3              | Endpoint: 'opc.tcp://FAGORCNC:4840'
14:49:45.037 | Server Node        | FagorOpcuaServer3              | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
14:49:45.037 | Server Node        | FagorOpcuaServer3              | ApplicationUri: 'http://www.FagorAutomation.com/machines/FagorCnc'
14:49:45.037 | Server Node        | FagorOpcuaServer3              | Used UserTokenType: UserName
14:49:47.379 | AddressSpaceModel  | FagorOpcuaServer3              | Registered for ModelChangeEvents
14:49:47.379 | Server Node        | FagorOpcuaServer3              | Connection status of server 'FagorOpcuaServer3' changed to 'Connected'.
14:49:47.379 | Server Node        | FagorOpcuaServer3              | Revised values: SessionTimeout=1200000, SecureChannelLifetime=600000
14:49:47.398 | AddressSpaceModel  | FagorOpcuaServer3              | Browse on node 'i=84' succeeded.
14:49:47.595 | AddressSpaceModel  | FagorOpcuaServer3              | Browse on node 'i=85' succeeded.
14:50:12.341 | AddressSpaceModel  | FagorOpcuaServer3              | Browse on node 'i=23470' succeeded.
15:05:31.065 | Server Node        | FagorOpcuaServer3              | Connection status of server 'FagorOpcuaServer3' changed to 'ConnectionWarningWatchdogTimeout'.
15:05:36.090 | Server Node        | FagorOpcuaServer3              | Connection status of server 'FagorOpcuaServer3' changed to 'Connected'.
15:11:40.063 | Server Node        | FagorOpcuaServer3              | Connection status of server 'FagorOpcuaServer3' changed to 'ConnectionWarningWatchdogTimeout'.
15:11:53.031 | Server Node        | FagorOpcuaServer3              | Connection status of server 'FagorOpcuaServer3' changed to 'Connected'.
```

Am I missing someting?
How did end up this issue?

Regards,
Ezio

NoelGraf, 2023-07-04:
@eziosella
It is important for which policy the server allows username/password authentication. On the server side, you can set a default access control using the UA_AccessControl_default function. In addition to the allowed usernames and passwords, the policyuri to use is also passed.
Take a look at the client_access_control and server_access_control examples.

eziosella, 2023-07-04:
@NoelGraf Thanks for the reply.

In the meantime I recompiled the project with ```-DUA_ENABLE_ENCRYPTION=OPENSSL``` cmake parameter,
then running the client_connect example I finally connected properly passing cert and key parameters as below:
```
./client_connect -username Admin -password myPassword -cert /home/ezio/open62541/tools/certs/server_cert.der -key /home/ezio/open62541/tools/certs/server_key.der -securityPolicy http://opcfoundation.org/UA/SecurityPolicy#None opc.tcp://FAGORCNC:4840
[2023-07-04 10:37:43.131 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-07-04 10:37:43.132 (UTC+0200)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2023-07-04 10:37:43.133 (UTC+0200)] info/securitypolicy        The basic256 security policy with openssl is added.
[2023-07-04 10:37:43.133 (UTC+0200)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2023-07-04 10:37:43.133 (UTC+0200)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2023-07-04 10:37:43.133 (UTC+0200)] info/eventloop     Starting the EventLoop
[2023-07-04 10:37:43.133 (UTC+0200)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-07-04 10:37:43.133 (UTC+0200)] info/network       TCP 4   | New connection to "95.244.143.134" on port 4840
[2023-07-04 10:37:43.300 (UTC+0200)] info/channel       TCP 4   | SC 11 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-07-04 10:37:43.300 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-07-04 10:37:43.802 (UTC+0200)] warn/client        The server returned Endpoints with a different EndpointUrl opc.tcp://FAGORCNC:4840 than was used to initialize the connection: opc.tcp://95.244.143.134:4840. Some servers require a complete match of the EndpointUrl/DiscoveryUrl (including the path) to return all endpoints.
[2023-07-04 10:37:43.802 (UTC+0200)] info/client        Selected endpoint 0 in URL opc.tcp://FAGORCNC:4840 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-07-04 10:37:43.802 (UTC+0200)] info/client        Selected UserTokenPolicy username_basic256 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2023-07-04 10:37:44.434 (UTC+0200)] info/securitypolicy        The basic256 security policy channel with openssl is created.
[2023-07-04 10:37:44.434 (UTC+0200)] info/securitypolicy        The basic256 security policy channel with openssl is deleted.
[2023-07-04 10:37:44.434 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-07-04 10:37:45.061 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-07-04 10:37:45.061 (UTC+0200)] info/userland      Connected!
[2023-07-04 10:37:45.100 (UTC+0200)] info/userland      The server date is: 01-07-2023 00:47:06.121
[2023-07-04 10:37:45.345 (UTC+0200)] info/channel       TCP 0   | SC 0  | Closed the SecureChannel
[2023-07-04 10:37:45.345 (UTC+0200)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2023-07-04 10:37:45.346 (UTC+0200)] info/network       TCP 4   | Socket closed
[2023-07-04 10:37:45.445 (UTC+0200)] info/eventloop     Stopping the EventLoop
[2023-07-04 10:37:45.445 (UTC+0200)] info/network       UDP     | Shutting down the ConnectionManager
[2023-07-04 10:37:45.445 (UTC+0200)] info/network       TCP     | Shutting down the ConnectionManager
[2023-07-04 10:37:45.445 (UTC+0200)] info/eventloop     The EventLoop has stopped
```

By my side problem solved!
Thanks,
Ezio

closed 2023-07-04 by NoelGraf