open62541/open62541 #5324: ua_pubsub.h: No such file or directory

shaohuazhang1, 2022-09-08:
[image]
[image]

Enable option UA_ENABLE_PUBSUB，detail see the picture，cmakeLists file option(UA_ENABLE_PUBSUB_MQTT "Enable publish/subscribe with mqtt (experimental)" ON)，off changed on；then //cmake .. //make //sudo make install；succeed,compile example tutorial_pubsub_mqtt_publish.c，
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include "ua_pubsub.h"
#include <open62541/plugin/pubsub_mqtt.h>
output as follow:
gcc -std=c99 tutorial_pubsub_mqtt_publish.c -o myServer
tutorial_pubsub_mqtt_publish.c:47:10: fatal error: ua_pubsub.h: No such file or directory
47 | #include "ua_pubsub.h"
| ^
compilation terminated.
Then I Enable UA_ENABLE_amalgamation,delete build document，recompile；succeed,tutorial_pubsub_mqtt_publish.c header filechanged“open62541.h",：
include "open62541.h"

include <signal.h>

compile gcc -std=c99 open62541.c tutorial_pubsub_mqtt_publish.c -o myServer
output as follow:
tutorial_pubsub_mqtt_publish.c: In function ‘main’:
tutorial_pubsub_mqtt_publish.c:463:53: warning: implicit declaration of function ‘UA_PubSubTransportLayerMQTT’; did you mean ‘UA_PubSubTransportLayerUDPMP’? [-Wimplicit-function-declaration]
  463 |     UA_ServerConfig_addPubSubTransportLayer(config, UA_PubSubTransportLayerMQTT());
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                     UA_PubSubTransportLayerUDPMP
tutorial_pubsub_mqtt_publish.c:463:53: error: incompatible type for argument 2 of ‘UA_ServerConfig_addPubSubTransportLayer’
  463 |     UA_ServerConfig_addPubSubTransportLayer(config, UA_PubSubTransportLayerMQTT());
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                     |
      |                                                     int
In file included from tutorial_pubsub_mqtt_publish.c:50:
open62541.h:37463:65: note: expected ‘UA_PubSubTransportLayer’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
37463 |                                         UA_PubSubTransportLayer pubsubTransportLayer);
      |                                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
tutorial_pubsub_mqtt_publish.c:475:5: error: unknown type name ‘UA_PubSubConnection’; did you mean ‘UA_PubSubConfiguration’?
  475 |     UA_PubSubConnection *connection = UA_PubSubConnection_findConnectionbyId(server, connectionIdent);
      |     ^~~~~~~~~~~~~~~~~~~
      |     UA_PubSubConfiguration
tutorial_pubsub_mqtt_publish.c:475:39: warning: implicit declaration of function ‘UA_PubSubConnection_findConnectionbyId’ [-Wimplicit-function-declaration]
  475 |     UA_PubSubConnection *connection = UA_PubSubConnection_findConnectionbyId(server, connectionIdent);
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
tutorial_pubsub_mqtt_publish.c:475:39: warning: initialization of ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
## Checklist

Please provide the following information:

 - [1.3.3 ] open62541 Version (release number or git tag):
 - [none ] Other OPC UA SDKs used (client or server): 
 - [ ubuntu 20.04] Operating system:
 - [ 300] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue
[image]

NoelGraf, 2022-09-08:
Your compiler settings are not complete. If you want to run the examples, I recommend enabling the CMake flag UA_BUILD_EXAMPLES. I can also advise you have a look at the [documentation](http://www.open62541.org/doc/master/building.html#building-the-examples)

shaohuazhang1, 2022-09-09:
Thank you!But when i enable example option,make ,new error occur: 
[image]
 make -j4
Scanning dependencies of target open62541-generator-types-powerlink
make[2]: *** No rule to make target '../deps/ua-nodeset/POWERLINK/Opc.Ua.POWERLINK.NodeIds.csv', needed by 'src_generated/open62541/types_powerlink_generated.c'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:2547: examples/nodeset/CMakeFiles/open62541-generator-types-powerlink.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target open62541-generator-transport
[  1%] Built target open62541-generator-statuscode
[  1%] Built target open62541-generator-types
make: *** [Makefile:152: all] Error 2

NoelGraf, 2022-09-09:
I created the project with your CMAKE setting and could not reproduce your error.
It would be best if you describe again in more detail how you build the project.
Also, I would recommend that you delete and recreate your build directory after changing the CMAKE configuration.

shaohuazhang1, 2022-09-13:
I delete the repository document open62541 before,execute git clone,build,make,error occur as follow,but I have installed openssl library.
usr/bin/ld: ../bin/libopen62541.so.1.3.3: undefined reference to `EVP_DigestVerifyUpdate'
/usr/bin/ld: ../bin/libopen62541.so.1.3.3: undefined reference to `EVP_PKEY_CTX_set_rsa_padding'
/usr/bin/ld: ../bin/libopen62541.so.1.3.3: undefined reference to `EVP_DigestSignUpdate'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/custom_datatype_server.dir/build.make:87: bin/examples/custom_datatype_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:887: examples/CMakeFiles/custom_datatype_server.dir/all] Error 2

> I created the project with your CMAKE setting and could not reproduce your error. It would be best if you describe again in more detail how you build the project. Also, I would recommend that you delete and recreate your build directory after changing the CMAKE configuration.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier