open62541/open62541 #5325: Could NOT find LATEX (missing: LATEX_COMPILER)

shaohuazhang1, 2022-09-09:
Description
When i execute configure on the cmake-gui ,error output as bellow:
bpf library not found. XDP functionalities will not be available.
Could NOT find LATEX (missing: LATEX_COMPILER) 
How to solve this problem?
Background Information / Reproduction Steps

Used CMake options:
[image]

 Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):1.3.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:Ubuntu20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

neojaw, 2022-09-09:
First off, the library compiles just fine without LaTeX, you just won't be able to build the documentation (which you arguably don't need to because you can find it online: [HTML](http://www.open62541.org/doc/1.3/toc.html), [PDF](https://open62541.org/doc/open62541-1.3.pdf)). So the line in the log is merely informational.

That being said, if you do want to build the documentation, you need to install LaTeX. On Ubuntu this should be some of the `texlive-something` packages. `texlive-latex-base` sounds like a good starting point. But be warned that the installation size is *huge*.

shaohuazhang1, 2022-09-09:
get
thank you!

closed 2022-09-09 by shaohuazhang1