open62541/open62541 #5328: Enable UA_BUILD_EXAMPLES error occured

shaohuazhang1, 2022-09-09:
Description
Error information
Scanning dependencies of target open62541-generator-types-powerlink
make[2]: *** No rule to make target '../deps/ua-nodeset/POWERLINK/Opc.Ua.POWERLINK.NodeIds.csv', needed by 'src_generated/open62541/types_powerlink_generated.c'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:2547: examples/nodeset/CMakeFiles/open62541-generator-types-powerlink.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target open62541-generator-transport
[  1%] Built target open62541-generator-statuscode
[  1%] Built target open62541-generator-types
make: *** [Makefile:152: all] Error 2

Background Information / Reproduction Steps

When i  Make after Enable EXAMPLE 

Used CMake options:

UA_BUILD_EXAMPLES

```bash
![Uploading cmake option.PNG…]()

```

Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):1.3.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:ubuntu20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

shaohuazhang1, 2022-09-09:
Enable UA_BUILD_EXAMPLES

marwinglaser, 2025-03-25:
We build examples in CI, this should be fixed.

closed 2025-03-25 by marwinglaser