open62541/open62541 #5332: OPC UA server connecting with real processs

alexmc1510, 2022-09-10:
Dear community,

First of all, many thanks for the enormous effort done behind the software. I am a newbie in that field, I am starting to play with the library and I am able to create my own OPC UA server. Now I would like to move one step further, and link the server with a real process. To do this I have a lot of devices in my lab, including: S7, Fanuc, OPC DA....but honestly, I don't know where to start. I would appreciate so much if you would be able to give me some examples of implementation of the link with one or some of that devices.

I know that this request is not an issue so feel free to move it to the corresponding folder or forum but I think this post will be very helpful for huge amount of people.

Thanks in advance,
Congrats

embbo, 2022-09-12:
YourDevice---OPCUA_Server----OPCUA_Client  and vice versa

alexmc1510, 2022-09-13:
Thanks for the quick answer. Do you have any example for the connection between "My device" and OPCUA_Server? Speaking about "My device" I have a lot of options: S7, Fanuc Focas, OPC DA, Modbus...
Honestly, I don't know how to create that link.

neojaw, 2022-09-13:
Well, "your device" probably has some kind of user interface where you can tell it where to connect to (the device where your server runs) and what to do with the connection (read/write variables etc). Check its manual, it's often a web interface that you can access with a browser.

The details really depend on your setup and your application.

embbo, 2022-09-13:
@alexmc1510 
just try establish your device and server .. my be without opcua server . you can find examples online .

alexmc1510, 2022-09-13:
But if my device is communicating over S7 of Fanuc FOCAS protocol, how can I covert that protocols into OPC UA?

neojaw, 2022-09-13:
If you want to connect something to an OPCUA server, it has to support OPCUA. I don't have any of your devices here, so I can only give you generic advice. Did you check the manual of your device(s)? Found a user interface?

alexmc1510, 2022-09-13:
But...I am missing something....as far as I understood, open62541 library allows you to create an OPC UA server, am I right? What I am asking is to link it with real data from PLC o NC.

neojaw, 2022-09-14:
> as far as I understood, open62541 library allows you to create an OPC UA server, am I right?

Right. And I thought you have already done that?

> What I am asking is to link it with real data from PLC o NC.

Well, first thing is to connect your PLC to the server you created. Your PLC, in this setup, acts as an OPCUA client. You need to tell it the network address of the server so it can connect to it. As I said, it probably has a user interface where you can do that.

As for exchanging data, that depends on what your PLC expects from the server. For example, a common way to exchange data is to read and write OPCUA variables on the server but then you might need to create those variables in your server code first. Or maybe your PLC wants to execute specific methods on the server, then you have to provide them. Your needed setup really depends on your PLC and your application here.

closed 2026-10-20 by o6-nbeier