open62541/open62541 #5333: QUESTIONS ABOUT COMMUNICATION SPEED

AlejandroAntonio1998, 2022-09-12:
## GENERAL INFO

cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=ON -DUA_MULTITHREADING=99 ..

 - [1.2 ] open62541 Version (release number or git tag):
 - Other OPC UA SDKs used (client or server): OPC UA SERVER WITH A PLC DEVICE
 - Operating system: LINUX

##DESCRIPTION
Hello, I am developing an application that communicates data between a PC linux (OPC UA client) and a PLC from Siemens (OPC server). At some point the PC needs to read data from 2000 nodes and it takes about 10-15 seconds. It means that the reading time of a single node is about 5 ms and I would need it to be faster.

The way I am reading the nodes is by using the function **UA_Client_readValueAttribute** for each node.

I have the following questions:
- Am I installing the library with cmake the wrong way in order to enable faster performance?
- Does it make sense the reading times that I have explained or should it be faster?
- Is it possible to usse a function that reads multiple nodes in a single action (using a different function or something like that)?

Thank you all for your time!

jpfr, 2022-09-12:
Yes. A single call can read multiple values.
Use the read service directly.

UA_Client_readValueAttribute Is a convenience method that abstracts the raw read service.

AlejandroAntonio1998, 2022-09-14:
Hello, I have tried the funtion **UA_Client_Service_read** and the reading times are similar.
Is it possible to configure the request or the client (or something) in order to make the communication faster?
It seems that each reading takes about 5 ms.
Thanks!

AlejandroAntonio1998, 2022-09-28:
I have decided to make an array of the data I want to communicate in order to reduce the ammount of nodes I have to read/write.
**Arrays** of 100 elements **can be read in a single call** of the function UA_Client_Service_read.
I hope this is helpfull to somebody!

closed 2022-09-28 by AlejandroAntonio1998