open62541/open62541 #5334: Learn open62541 - subscribe using mqtt

jhsiemens, 2022-09-13:
## Description
I currently try to implement example [https://chowdera.com/2022/150/202205301114102452.html](https://chowdera.com/2022/150/202205301114102452.html) to print mqtt publishing data to console of subscriber.

## Background Information / Reproduction Steps

I can see the publish messages in local mqtt broker and publisher and subscriber are successful build:

[image]
[image]

all is working through by checking with UA_LOGCATEGORY_SERVER
only the method callback is not called, which should be done with:
    **UA_StatusCode rv = connection->channel->regist(connection->channel, &transportSettings, &callback);**

**Used CMake options:**

UA_ENABLE_DA UA_ENDABLE_JSON_ENCODING UA_ENABLE_METHODCALLS UA_ENABLE_PUBSUB UA_ENABLE_SUBSCRIBTIONS UA_ENABLE_SUBSCRIBTION_EVENTS UA_FORCE_WERRORS

Include all CMake options here, which you modified or used for your build.

**If you are using cmake-gui, go to "Tools > Show my Changes" and paste the content of "Command Line Options"**

Commandline options:
-DUA_ENABLE_ENCRYPTION_TPM2:STRING="OFF" 

Cache file:
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF

On the command line use `cmake -L` (or `cmake -LA` if you changed advanced variables)
-->

```bash
cmake -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2022-09-13:
If you followed the tutorial, you are probably on the 1.3 branch. I would recommend you switch to the master branch. There some changes have been made regarding pubsub mqtt. There is also a tutorial_pubsub_mqtt_subscribe example.

jhsiemens, 2022-09-13:
Following executables where created and placed in folder open62451/ after cmake build with described options
[PubSubPrintConsole.zip](https://github.com/open62541/open62541/files/9557051/PubSubPrintConsole.zip)