open62541/open62541 #5339: I wonder if this supports reverse connection.

icozzz, 2022-09-17:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-09-21:
Not at this time.
But it is actively worked on by the company Basyskom.
Give it a few months.

basyskom-meerkoetter, 2022-10-26:
https://github.com/open62541/open62541/pull/5363

closed 2024-05-26 by icozzz